Ultimaker/Cura #1774: Estimated print time always too short

probonopd, 2017-05-01:
The estimated print time in Cura is always too short for my printer (Renkforce RF100). I suspect this has to do with the fact that the acceleration and print speeds are not taken into the calculation correctly. The printer gives

```
echo:Marlin 1.0.3 dev
echo: Last Updated: Nov 18 2016 13:36:28 | Author: (none, default config)
Compiled: Nov 18 2016
echo: Free Memory: 4436  PlannerBufferBytes: 308
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X78.82 Y78.82 Z200.00 E106.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X500 Y500 Z100 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P500.00 R3000.00 T500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 M0 H220 B70 F0
echo:  M145 M1 H240 B110 F0
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00 C1.00
echo:Filament settings: Disabled
echo:SD init fail
```

Do I need to adjust the printer definition get more accurate time estimations?

nallath, 2017-05-02:
As far as I know cura uses the accelerations that it has to calculate the speed. You could try changing the settings in Cura.

filipgoc, 2017-05-02:
just flagging this for myself.in case you do experiment with printer definitions and estimation accuracy (more / less overrides in the json), then I'm curious.

probonopd, 2017-05-03:
@nallath thanks for the hint, I am not trying to optimize this on my own machine but rather to improve the printer definition file. Do you have any hint for me which keys might have wrong or missing values? Thanks.

Ghostkeeper, 2017-05-05:
First take a look at the jerk and acceleration settings, I'd say. That is `acceleration_enabled` but also `acceleration_print`, `acceleration_infill`, etc. And the same for `jerk_*`.

BagelOrb, 2017-07-21:
There are machine settings to notify Cura of the firmware settings the printer has without changing the acceleration etc.

See the settings starting from https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json#L463

closed 2017-07-21 by BagelOrb