Ultimaker/Cura #1775: Ultimaker 3 using both extruders when only 1 material is required

fish0and0chips, 2017-05-02:
When setting up a print to use just extruder 1 or extruder 2 (we have PLA in 1 and PC in 2), Cura automatically sets up the print to print the first layer with one material and then continue with the other. This is fixed by turning on support material, selecting the correct material and then switching off support again (as we are not using support in this print). This is frustrating to go through every time we switch between the two materials.

You can see in the attached picture that the path of the extruder shows it priming each head in the corner.

[image]

Ghostkeeper, 2017-05-05:
Ah, huh. We recently had this bug. It seems it was only fixed in the front-end, not in the back-end.

Ghostkeeper, 2017-05-05:
Hmm, I can't reproduce this bug. Perhaps you could export your set-up as a project and upload it here, to see if we can reproduce it?

BagelOrb, 2017-07-21:
Which version of Cura are you using?

fish0and0chips, 2017-07-21:
Hi, sorry, this was in an old build of Cura and doesn't seem to be a problem any more.

closed 2017-07-21 by fish0and0chips