Ultimaker/Cura #1777: USB printing - Not working for Smoothieware firmware

Shaische, 2017-05-02:
There is an issue with printing via USB when attempting to print with a smoothieware firmware based printer. This issue didn't exist in Cura 1, so it would be good to compare the code to see what changed that is causing this.

The original problem with Cura 2 was not being able to communicate with the printer at all and that was due to it sending empty command lines to the printer. Clicking "Print via USB" did not do anything at all. Jaime had been able to fix this, but not sure if it was merged into 2.5.0. 

After testing a nightly build that had the fix of not sending empty command lines, we were able to start the print. But, it would only get to the point where the hotend heated up, moved down to the bed to start to print and would only deposit maybe about 2mm worth of plastic before the printer froze in place and wouldn't continue further.

If anyone can fix this, we would be very grateful! We can test it for you.

nallath, 2017-05-03:
I've made a few changes outside of working hours, but as I only have a board without a full printer attached, it's hard to see what's going wrong. 

The code difference between Cura 1 and Cura 2 are pretty big. Also considering that cura 1 used python 2, whereas cura 2 used Python 3. The USB code of cura2 is based on that of the old one, but it's impossible to completely duplicate it.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Shaische, 2017-11-14:
Yes, for smoothieware it is still an issue.

nallath, 2017-11-15:
As Ultimaker doesn't use smoothieware, i don't think its likely that anyone from Ultimaker is going to spend time on fixing it. If someone from the comunity fixes it, we're more than happy to merge the pull request.

PR #3346 (2018-03-14): Various bug fixes for USB/serial attached printers
[edit]
I might have foobared this PR, thought I could send it as a separate, but I might have missed something. Please let me know if I should submit it in another way.

This PR contains various fixes in handling of USB/Serial printers, see commit messages for more info. With these fixes Cura should play nice with these printers. Tested on a Cetus3D with TinyFab smoothieware controller. 
[/edit]

Do not send new G-CODEs when the RX buffer is filling up. This is in … an attempt to fix #1777 where smoothieware based printers would halt in the mi [... cut]
commit 112bb260 (2018-03-14): Do not send new G-CODEs when the RX buffer is filling up. This is in an attempt to fix #1777 where smoothieware based printers would halt in the middle of the print.
commit a024be78 (2018-03-14): Reverting this commit as this was not the correct way to fix it.
Revert "Do not send new G-CODEs when the RX buffer is filling up. This is in an attempt to fix #1777 where smoothieware based printers would halt in the middle of the print."

This reverts commit 112bb260acb8655e929cf377b18f505d62608eb1.
closed 2018-03-14 by Ghostkeeper