Ultimaker/Cura #1779: License of fdm_materials

onitake, 2017-05-02:
The fdm_materials repo does not contain any licensing information.

Since the material definitions are an integral part of Cura, one would assume they are released under the same license (AGPL-3+).
Can you clarify if this is the case, and maybe add a LICENSE file so there is no ambiguity?

Thank you!

nallath, 2017-05-03:
They are data files, so I'm not quite sure if you could even put a licence on it if you wanted it.

hroncok, 2017-05-03:
> They are data files, so I'm not quite sure if you could even put a licence on it if you wanted it.

OF course you could. Data can has its license too.

nallath, 2017-05-03:
No, something can only have a licence if you can actually exert any rights on it. Source code can have a licence as you can have rights on it (as it's seen the same as writing a book). 
As far as I understood it, data files don't fall under that branch of copyrights.

fieldOfView, 2017-05-03:
Someone wrote the data, or the script that produced the data from other data. That person has rights to the data. A printed catalog of materials is also a book.

hroncok, 2017-05-03:
There are even [open licenses for data](https://opendatacommons.org/licenses/).

nallath, 2017-05-03:
Part of the files have been auto generated. So you can have copyright on the script that produced them, but that does not imply that the result is also copyrighted (or copyrightable). If it would be, everything created with Cura would be AGPL (and i doubt that g-codes, which Cura produces, are copyrighted!).

The call isn't up to me, but I don't think that adding any licence here makes sense or adds anything.

hroncok, 2017-05-03:
Well as the data are now in https://github.com/Ultimaker/fdm_materials they should be considered non-redistributable and non-free as they have no license information. As such, I for example cannot package it for Fedora. I guess the same thing applies for Debian and others.

hroncok, 2017-05-03:
> Part of the files have been auto generated. So you can have copyright on the script that produced them, but that does not imply that the result is also copyrighted (or copyrightable). If it would [...]

I agree with that. A license of a program does not imply the license of the generated data. This does not mean the data cannot be licensed by the person or company who generated them.

> If you read their FAQ, you can already see the difficulty that they have with such a thing. They also describe cases like databases with photos or technical designs. It's not such a simple matter.

Licenses are always difficult :)

nallath, 2017-05-03:
My interpretation is that these files are configuration files. Software holds rights because it is seen as an "expression". These files only hold "factual" data and can not be quite seen as a form of expression. One might argue that the comments in the file are copyrightable, but as far as I know, we hardly have any of those.

This would mean that no-one can derive any rights over these files; so are therefore distributable and free. 

If it's really required, Ultimaker could put a license on this, but that would have about as much value as me handing out licenses for the usage of air.

hroncok, 2017-05-03:
You could put a public domain on that if tou feel that's basically the case anyway.

onitake, 2017-05-03:
The issue arises here, because the configuration files are distributed as part of a larger application.

In any case, I think it wouldn't hurt to add proper copyright and a license. But it would avoid all ambiguity.

onitake, 2017-06-14:
Is there any new information from Ultimaker with regards to the fdm_materials license?

I'd like to finalise the Cura 2.5 Debian packages before 2.6 is out, and having this sorted is a small but crucial piece for the release process.

If Ultimaker is ok with making the material definitions available for any use, a short notice like "These files are released into the public domain" or similar should suffice. That way, the intent is clear.

nallath, 2017-06-14:
I just discussed this with management. All files in this repository can be seen as part of the public domain. Anyone is free to use them in any way they want.

hroncok, 2017-06-14:
Thanks @nallath.

onitake, 2017-06-14:
Thank you very much, @nallath !
I'm going to close the issue now.

closed 2017-06-14 by onitake

nickthetait, 2017-06-19:
Hooray for the decision. However, I do not consider this issue to be resolved yet as there is not actually a license in that repo. Please commit one.

Recommend using [CC by 4.0 ](https://creativecommons.org/licenses/by/4.0/) which allows:

> Share — copy and redistribute the material in any medium or format
> Adapt — remix, transform, and build upon the material for any purpose, even commercially.

fieldOfView, 2017-06-20:
Other options are CC0 and Unlicense:
https://choosealicense.com/licenses/cc0-1.0/
https://choosealicense.com/licenses/unlicense/

Ghostkeeper, 2017-06-27:
I found a workaround, sort of. This repository does have a CMakeLists.txt file, which is a creative expression.

I've added a CC0 license to this repository. According to Nallath this license only applies to the CMakeLists.txt file. According to Hroncok this also applies to the data files. Whatever is the case, the data files can be used for whatever purpose you like.