Ultimaker/Cura #1784: Desktop file invalid

hroncok, 2017-05-03:
Application Version: 2.5.0
Platform: Linux (Fedora 27 to be precise, but this would happen on other Linuxes as well)
Qt: not relevant
PyQt: not relevant
Display Driver: not relevant

Steps to Reproduce:

    cmake ...
    make
    cmake install ...
    desktop-file-validate /usr/share/applications/cura.desktop

Actual Results:

    /usr/share/applications/cura.desktop: error: value "1" for key "Version" in group "Desktop Entry" is not a known version

Expected results:

    (empty)

Additional Information:
The [latest specification](https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-latest.html) says:

> Version of the Desktop Entry Specification that the desktop entry conforms with. Entries that confirm with this version of the specification should use 1.1. Note that the version field is not required to be present.

closed 2017-05-03 by awhiemstra

hroncok, 2017-05-03:
Thanks for fixing.

probonopd, 2017-08-18:
Still invalid in 2.6.2:

```
cura.desktop: error: (will be fatal in the future): value "cura-icon.png" for key "Icon" in group "Desktop Entry" is an icon name with an extension, but there should be no extension as described in the Icon Theme Specification if the value is not an absolute path
```
__Solution:__ Remove `.png`

```
cura.desktop: error: value "application/sla;application/vnd.ms-3mfdocument;application/prs.wavefront-obj;image/bmp;image/gif;image/jpeg;image/png" for string list key "MimeType" in group "Desktop Entry" does not have a semicolon (';') as trailing character
```
__Solution:__ Add `;`
Reference:
https://travis-ci.org/AppImage/AppImageHub/builds/266072586#L538-L539

commit cee05391 (2017-05-03): Remove version field from Desktop file

Apparently, it is not required and we are specifying an invalid version.

Fixes #1784
commit b9970d57 (2017-08-30): Make it pass desktop-file-validate, closes  #1784

```
cura.desktop: error: (will be fatal in the future): value "cura-icon.png" for key "Icon" in group "Desktop Entry" is an icon name with an extension, but there should be no extension as described in the Icon Theme Specification if the value is not an absolute path
```
__Solution:__ Remove `.png`. __PLEASE NOTE__ that for this to work, cura-icon.png must be located in one of the directories in which desktops search for icons, such as `/usr/share/icons/hicolor/128x128/apps/`. Please make sure it ends up there (also in the AppImage, too.)

``` [... cut]