Ultimaker/Cura #1789: Automatically connects to the printer

kr15uk, 2017-05-04:
Ladies and gents,

This is more of a feature request I'd say - is it possible to make little toggle switch or tick box in settings so Cura doesn't connect to the printer automatically when machine gets plugged in?!

I do have Cura running in the background pretty much all the time and for example if I quickly need to upload some changes to the machine via Arduino IDE then I need to quit Cura 1st - which is a bit of an inconvenience.

Thanks.

Kris

awhiemstra, 2017-05-04:
We have an open PR from @fieldOfView that reworks the way we deal with USB connections. See https://github.com/Ultimaker/Cura/pull/1688 . This should solve the auto-connect problem.

closed 2017-05-04 by awhiemstra

fieldOfView, 2017-05-04:
Note that I don't think that PR will be ready to be merged in time for 2.6