Ultimaker/Cura #1790: Cura 2.5.0 view GCODE

kr15uk, 2017-05-04:
Ladies and gents,

1st - I love this new feature of v2.5.0 but would be nice to see some extra features:
1. See a bit more details like tepms and the layer height.
2. Be able to tweak basic settings like temperature (hotted and bed).

Maybe only for Cura generated GCODE files so there's no confusion like with variable layer height from different slicers or so...

Probably low priority but would be nice to do that on the spot.

Thanks.

Kris

Ghostkeeper, 2017-05-05:
We've had longstanding plans to allow modifying the g-code afterwards, with an editor like Repetier-Host has. This has virtually no priority over other plans though.

ianpaschal, 2018-05-29:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

Sorry about that!