Ultimaker/Cura #1791: Image conversion Cura 2.5 OSX 10.12

megrimm, 2017-05-04:
Hello,

When importing an image for conversion for pen plotting on a 3d printer Cura crashes (spinning beach ball). I tried importing a .jpg and a .png:

height = 0.2
base = 0
width = 80
depth = 28.887949260042284

On Cura 1.3.1 I used to just get "can not import" error.

Thanks

Vandrasc, 2017-05-05:
I could reproduce but only when I am using a comma instead of a decimal point. Is that your case as well? Thanx! 
Dev see CURA-3754 for more info

megrimm, 2017-05-05:
hmmmm... no decimal is there.

also, 3d viewing area goes blank. i tried a different .jpg and get the "failed to load" error.

does it matter what size the image is? other necessary image specifications?

Ghostkeeper, 2017-05-05:
The image gets scaled to a limit of 512x512, iirc. It should be able to handle all image sizes as long as they fit into memory. You just won't get all the details of very big pictures.

Perhaps you could share a log with us? Then we can debug this.

megrimm, 2017-05-05:
sure. log attached.
[cura-log.zip](https://github.com/Ultimaker/Cura/files/979116/cura-log.zip)

Ghostkeeper, 2017-05-10:
For the lazy, I see these two relevant errors in the log. This one is in there twice:

    2017-05-05 09:11:55,451 - ERROR - UM.Logger.logException [74]: Exception: Job <UM.Mesh.ReadMeshJob.ReadMeshJob object at 0x600012a677d8> caused an exception
    2017-05-05 09:11:55,453 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
    2017-05-05 09:11:55,455 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 122, in run
    2017-05-05 09:11:55,456 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/ReadMeshJob.py", line 39, in run
    2017-05-05 09:11:55,458 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 581, in getBoundingBox
    2017-05-05 09:11:55,459 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Scene/SceneNode.py", line 648, in _calculateAABB
    2017-05-05 09:11:55,460 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 176, in getExtents
    2017-05-05 09:11:55,465 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 251, in getConvexHullVertices
    2017-05-05 09:11:55,467 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 243, in getConvexHull
    2017-05-05 09:11:55,468 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 235, in _computeConvexHull
    2017-05-05 09:11:55,470 - ERROR - UM.Logger.logException [78]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Mesh/MeshData.py", line 365, in approximateConvexHull
    2017-05-05 09:11:55,472 - ERROR - UM.Logger.logException [78]:   File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21663)
    2017-05-05 09:11:55,473 - ERROR - UM.Logger.logException [78]:   File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3879)
    2017-05-05 09:11:55,474 - ERROR - UM.Logger.logException [78]: scipy.spatial.qhull.QhullError: Qhull error

And this error gets repeated tons of times.

    2017-05-05 09:11:56,780 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
    2017-05-05 09:11:56,782 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
    2017-05-05 09:11:56,783 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Qt/Bindings/MainWindow.py", line 184, in _render
    2017-05-05 09:11:56,784 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/Qt/QtRenderer.py", line 138, in render
    2017-05-05 09:11:56,786 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/View/SelectionPass.py", line 67, in render
    2017-05-05 09:11:56,787 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/View/SelectionPass.py", line 104, in _getNodeColor
    2017-05-05 09:11:56,789 - CRITICAL - cura.CrashHandler.show [39]:   File "/Users/ultimaker/build/2.5/build/inst/lib/python3.5/site-packages/UM/View/SelectionPass.py", line 123, in _isInSelectedGroup
    2017-05-05 09:11:56,790 - CRITICAL - cura.CrashHandler.show [39]: AttributeError: 'NoneType' object has no attribute 'callDecoration'

Ghostkeeper, 2017-05-10:
Still cannot reproduce :(

BagelOrb, 2017-07-21:
What images are you using?
At which resolutions are the settings with which you are trying to import it?

ianpaschal, 2018-02-27:
Was never possible to reproduce, and labeled `Won't Fix` in JIRA, so I'm closing.

closed 2018-02-27 by ianpaschal