Ultimaker/Cura #1792: Cura 2.3.1 The program does not start, the error occurs at the stage of "updating the configuration"

extrim558, 2017-05-04:
The program does not start, the error occurs at the stage of "updating the configuration"

Version: 2.3.1
Platform: Windows-Vista-6.0.6002-SP2
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

fieldOfView, 2017-05-04:
Remove the file ```C:\Users\[yourname]\AppData\Local\cura\cura.log```

fieldOfView, 2017-05-04:
You may also want to update to Cura 2.5. Lots has been improved since Cura 2.3.1

extrim558, 2017-05-04:
I can not find the file cura.log
I have 32 bit OS. I would happily switch to version 2.5

extrim558, 2017-05-04:
Through the explorer, I could not find the path to this file. I wrote this path in the search bar and I was able to find the file cura.log. Thank you. Cura working!

closed 2017-05-04 by extrim558