Ultimaker/Cura #1793: Cura 2.3.1 can not rotate the model along the axis X and Y

extrim558, 2017-05-04:
Cura 2.3.1 can not rotate the model along the X and Y axes. Rotation of the model can be carried out only along the Z axis. When I move the mouse cursor to the rotation strip of the X and Y axis, they are not even active. Only the Z-axis band is active. It is very uncomfortable.

Ghostkeeper, 2017-05-05:
Is that better in Cura 2.5?

Though I don't remember changing anything related to that since 2.3.1.

extrim558, 2017-05-05:
I do not speak English.
I did not understand your idea.
watch the video https://youtube.com/watch?v=N1jKl7I-Eqo

Ghostkeeper, 2017-05-10:
Update Cura 2.3.1 -> Cura 2.5. Try again.

extrim558, 2017-05-10:
I have Windows 32bit

Tangenter, 2017-07-30:
I have exactly the same problem like extrim558. Cura can not rotate the model along the Y and Z axes. Only the "red" X-axle works. 
I have the problem with Cura 2.5. and even 2.6. 
Only version 15.4. is fully functional.
The issue ocures with several objects of thingiverse and even with my own designs made with Inventor.

My system is Windows 7 64bit.

Does anyone has a solution?

fieldOfView, 2017-07-30:
Is it just rotation or do the other tools (move, scale, mirror) behave similarly?

Tangenter, 2017-07-30:
You are right, the other tools (move, scale, mirror) behave similarly. I only recognized it at the rotation

Tangenter, 2017-08-02:
Does anyone have a guess?

Ghostkeeper, 2017-08-14:
I honestly have no idea. But I'm wondering if there is the "highlight" effect when you hover over the tool handles. If you go to Solid view (not layer view or X-ray), and you move your mouse over the tool handles for rotating in the green or blue direction, does it highlight the tool handle? It is a very slight effect, so look closely!

If it doesn't even hover then, that could indicate that there is a rendering bug going on on your computer and we'll need to know what video card and driver you're using. Basically what happens is that we draw the scene in the background and give the tool handles a unique colour, and then sample the pixel underneath the cursor to know which objects to highlight and which to select when you click. This could be a bit fragile on some computers, so maybe that broke for you.

Tangenter, 2017-08-16:
There is only a "highlight" effect for the red x axle. The Green and the Blue lines don't shine.

Thank you for your time

Ghostkeeper, 2017-08-29:
So the rendering seems to be fine then :confused:

Tangenter, 2017-08-29:
Hmmm... I think I have to look for a different Slicer. Thank you for your help

awhiemstra, 2017-08-30:
Note that you did not mention what platform and video card you are using. Since I suspect this issue is related to your video card drivers, would you mind sharing that information?

Tangenter, 2017-08-31:
System: Windows 7 Professional Service Pack 1
Processor: AMD Phenom II X4 945 3.00 GHz
Video card: ATI Radeon HD 4800 Series / driver version 8.632.1.2000

Tangenter, 2017-09-13:
the problem is solved. a new installation of the graphic driver has helped. Thanks to all

closed 2017-09-13 by nallath