Ultimaker/Cura #1795: Cura 2.5 included plugins not available to use

bapftr, 2017-05-05:
Cura 2.5 (Downloaded today 5/4/2017) only shows plugins for:
Pause at Height (BQ Printers)
Pause at Height
Pause at Height for repetier
TweakAtZ 5.1 (Experimental)
in the Post Processing Scripts menu.
However the included plugin list in Cura's configuration shows many more.
But they are not accessible to use, only the above listed 4.

BAP

fieldOfView, 2017-05-05:
You are confusing post-processing _scripts_ with modular cura _plugins_.

Cura 2.x has been rewritten from the ground up to be modular. Just about any functionality in Cura is a plugin: from the different view modes (solid, xray, layers) to support for file formats (STL, 3MF, etc), connections with USB printers and networked connections to The Ultimaker 3 and OctoPrint, everything is a _plugin_. These are the plugins you see listed in the preference pane.

One of the plugins is the Post Processing _plugin_, which allows you to run _scripts_ to alter the gcode after it has been produced by CuraEngine. In the old Cura, these scripts were named plugins, which is probably the cause of your confusion.

closed 2017-05-08 by nallath

apphead, 2017-05-14:
Naming confusion aside, is there any information available on how to create and implement post-processing _scripts_ in Cura 2.5+ ? The Post Processing _plugin_ is not even available on github/plugins/

That would greatly facilitate additional functionality be programmed by the community...

fieldOfView, 2017-05-14:
There's only learning by example, I am afraid. PostProcessingPlugin specifically comes with an example plugin.

bapftr, 2017-05-16:
Thanks for the explanation, I understand it better now.