Ultimaker/Cura #1798: Cura places my print below the bedplate.

MarkAtZahners, 2017-05-05:
Cure keeps putting different prints of mine below the buildplate. Even after i hit lay flat. Multiple versions of cura have done this and on many different prints. Any ideas?

Thanks

[image]

Ghostkeeper, 2017-05-10:
I have no clue. It doesn't happen for the models I tried. How can we reproduce this? When does it happen? With all the models you try?

ChrisTerBeke, 2017-11-14:
Is this still an issue?

MarkAtZahners, 2017-11-14:
Not I think it was a bug from like 5 versions ago

Thanks

-- 

[image: www.zahnerinc.com] <http://www.zahnerinc.com>

*Mark P. Prost *| Zahner & Associates, Inc.

Laser Scanning and 3D Printing Specialist

200 Zahner Place, Perryville, MO 63775

t. 573.547.1771| c. 573.768.4675 | f. 573.547.1452  | Zahnerinc.com

*Click here* <http://www.zahnerinc.com/laser.html>* for information on
Mobile LiDar & Scanning Services*

*[image: Zahner Inc. Facebook] <http://www.facebook.com/zahnerinc>*

CONFIDENTIALITY NOTICE: The contents of this email message and any
attachments are intended solely for the addressee(s) and may contain
confidential and/or privileged information and may be legally protected
from disclosure. If you are not the intended recipient of this message or
their agent, or if this message has been addressed to you in error, please
immediately alert the sender by reply email and then delete this message
and any attachments. If you are not the intended recipient, you are hereby
notified that any use, dissemination, copying, or storage of this message
or its attachments is strictly prohibited

closed 2017-11-15 by nallath

Ghostkeeper, 2017-11-24:
We discovered that this bug was still in the current release. We made a fix yesterday: https://github.com/Ultimaker/Uranium/commit/e617dd810585e49f16f0e12d10318713856ccaae