Ultimaker/Cura #1801: Center model function resets orientation as well

dtwrv6, 2017-05-06:
Windows version - same bug in 2.5 and 2.4

After rotating a model and positioning it, select "Center model on platform".  Not only does it center the model, it resets all orientation changes.  This should be a separate function.

nallath, 2017-05-08:
I can't reproduce this. Center model only changes the position for me.

Ghostkeeper, 2017-05-10:
Perhaps it only happens with grouped models or something?

BagelOrb, 2017-07-21:
I can reproduce a bug (2.6 on Windows) which is liklely to be what the OP is experiencing.

Try out some weird orientations and try using the mirror tool. Then centering also does some flipping or rotating by 180 degrees.

ChrisTerBeke, 2018-01-03:
Fixed in https://github.com/Ultimaker/Cura/commit/e0c2a9e4c7780addc043859369edad400b63a1b0.

closed 2018-01-03 by ChrisTerBeke