Ultimaker/Cura #1803: Cura support gcode has huge extrusion commands

LeviSchuck, 2017-05-07:
On Mac, Cura 2.5 will insert weird huge extrusion commands around points

To remove these and have a successful print, I had to 
`grep '^G1 F[0-9]* E[0-9.]*$' LY_gp4s-0v2.gcode > LY_gp4s-regexed.gcode`
This leaves the negative extrusion (retractions) alone.

```
G1 F1500 E3858.28469
...
G1 F1500 E4512.12118
...
G1 F1500 E4518.62118
```
(lines not necessarily directly after another)

Interestingly, for a large portion of the file, the extrusion amount has an average upward trend, this is during the main print, not start, skirt, or end. Is this supposed to be reporting some sort of progress?

This causes the printer to stop, extrude the specified amount, and continue.
However, extruding that crazy amount in the middle of the print makes all prints in the same area fault and then print spaghetti.

Attached is the gcode generated from Cura, as well as the original STL which can be found [on thingaverse](http://www.thingiverse.com/thing:1640681)
[gopromount.tar.gz](https://github.com/Ultimaker/Cura/files/981300/gopromount.tar.gz)

Again, this seems to be primarily but not unique to using support.
```
G0 X19.014 Y1.735 Z6.3
;TYPE:SUPPORT
G1 F1500 E1329.09904
G1 F4800 X19.05 Y1.784 E1329.10107
```

It will still add arbitrary globs in the beginning too, whether or not the model includes support.

smartavionics, 2017-05-07:
Here's part of your gcode that includes one of the extrusions you cite above:

G1 X-22.486 Y1.6 E3858.17706
G1 X-22.486 Y-1.6 E3858.2835
G1 X-22.45 Y-1.6 E3858.28469 <---------- last extrusion in sequence before move
G1 F1500 E3851.78469  <----------- retract
G0 F7200 X-22.036 Y-2.398 <---- move sequence
G0 X-22.036 Y-3.277
G0 X-20.977 Y-4.007
;TYPE:FILL
G1 F1500 E3858.28469    <---------- prime to return to previous extruder position
G1 F4800 X-21.889 Y-4.918 E3858.32757
G1 X-21.959 Y-4.988
G0 F7200 X-21.95 Y-4.988

It's simply undoing the previous retraction, isn't it?
Cura uses absolute extrusion mode, not relative mode like some slicers do.

LeviSchuck, 2017-05-07:
Very interesting.
So I take it that since I had to remove these lines to get an acceptable print, but the other absolute extrusion commands work fine, that there is a bug in the firmware on my printer?

smartavionics, 2017-05-07:
Hi, it's doing something like 6.5mm of retraction. That would be appropriate for a printer with a reasonably long bowden tube. But perhaps it's not right for your printer. Do you have a bowden extruder? I believe that 1% of the bowden length + 1mm is a good initial retract length which can then be fine tuned. If you have a direct extruder 6.5mm is going to be way to much retraction. Perhaps you want to play with the retraction settings?

LeviSchuck, 2017-05-09:
@smartavionics  Looks like you were spot on.

I did the math you did on one of the example code files and it works with 0.5mm retraction it seems.
I replicated it into my settings, the hidden default in Cura was 6.5mm which is where this came from!

Indeed my printer uses a bowden tube.

Again, thanks for your time and help understanding.

closed 2017-05-09 by LeviSchuck