Ultimaker/Cura #1804: cura2.5 Funnel support is sometimes vacant

651851996, 2017-05-07:
cura2.3 Also have this problem，and 30° have this problem，20°have not

Ghostkeeper, 2017-05-11:
Funnel support... You mean conical support?

Seems to work here.
[image]

BagelOrb, 2017-07-21:
Could you please explain the problem a bit more?
I currently don't understand what the problem is.

651851996, 2017-07-21:
emm,use 30°,Wide range of support，Sometimes slashes are lost。Generally occurs in the oblique line and support the same---using Google Translate

BagelOrb, 2017-07-21:
Could you attach screenshots?

Which language do you speak? Perhaps we can understand that better than Google translate can translate it ;)

651851996, 2017-07-21:
有道理。就是用30°Funnel support，当Funnel的方向和support的走向平行，可能会出现部分support丢失并产生问题

BagelOrb, 2017-07-21:
Please include pictures of the layer view.

Please include the ".curaproject.3mf" file.

651851996, 2017-07-21:
now without pc，wait some day┐(´д｀)┌

BagelOrb, 2017-07-21:
Ah ok.

651851996, 2017-07-23:
looklike that
[image]
[image]
[image]

Ghostkeeper, 2017-08-01:
I could reproduce this. It only seems to happen with zigzag support. What happens is well visible in 651851996's last screenshot: That stray line that's encircled **almost** hits the normal zig-line to the left of it, but not quite. On the layer below, the conical support pushes the line more to the left and as a result the line is now part of the normal border of the support area and because this line is on the opposite side of the zag, the line is not drawn any more.

I wonder what causes the loose lines like this though.
[image]

651851996, 2017-08-13:
Is this model

651851996, 2017-08-13:
Is this model
[image]

Ghostkeeper, 2017-10-30:
Our project manager just removed this from our planning because it's 12 weeks old.

Ghostkeeper, 2018-03-13:
Yes, it still occurs. Nothing has changed in this regard.

[image]

closed 2023-06-24 by github-actions[bot]