Ultimaker/Cura #1807: Cura 2.5 save dual extruder profiles

lokanaft, 2017-05-08:
It's can't save different settings in different exruders. After save 1 and 2 are merged.

nallath, 2017-05-08:
Are the settings you tried to save global settings? Could you share a better set of expected behavior vs actual behavior?

lokanaft, 2017-05-08:
My settings are different:

[image]

[image]

Then I save it:

[image]

And first extruder becoming as second:

[image]

Ghostkeeper, 2017-05-11:
I'm unable to reproduce this in 2.6. We changed quite a lot of things around for the extruder stacks in 2.6 so maybe this was accidentally fixed in the transition.

lokanaft, 2017-06-30:
Installed version 2.6.1 - everything repeats, the problem is not fixed.

nallath, 2017-06-30:
Could you share your local setup folder? We're having troubles reproducing this.

lokanaft, 2017-06-30:
Put folders in zip? What folders I need put?

Ghostkeeper, 2017-07-06:
In Cura, go to "Help" -> "Show configuration folder". The folder "2.6" you see there is what we'd like to see. It basically contains your entire configuration.

Also, deleting that folder restores Cura to factory defaults.

lokanaft, 2017-07-06:
Here: [2.6.zip](https://github.com/Ultimaker/Cura/files/1127432/2.6.zip)

nallath, 2017-07-06:
You seem to be using a machine definition that i've never seen. How did you get the "fabrikator" machine definition?

nallath, 2017-07-06:
Okay, so i'm guessing that something went wrong while doing that.

Ghostkeeper, 2017-07-09:
It's not just Cura's cloning routine. This one actually came from a definition called `fabrikator_dual`.

lokanaft, 2017-07-13:
Ok, now create clean dual extruder printer and all repeat: [2.6.zip](https://github.com/Ultimaker/Cura/files/1144387/2.6.zip)

Ghostkeeper, 2017-07-17:
I can see it happening now as well, but it only happens when updating the current profile, not when saving it to a new profile.

Ghostkeeper, 2017-07-17:
The question now becomes, how did lokanaft get his Cura into that state? If I try to reproduce it normally it seems fine. I also can't find any mistakes in the actual settings files in the .zip (though it must be in there somewhere).

lokanaft, 2017-07-17:
What I need do now?

Ghostkeeper, 2017-07-17:
Well, nothing, the ball is in our court, so to say :wink: Unless you remember the sequence of steps that got this printer in Cura. It's not happening when I try to reproduce it from a clean install.

Ghostkeeper, 2017-07-24:
Still unable to reproduce this from our end though. It only happens with the project file Iokanaft provided.

Perhaps some more information then. You said you created a clean dual extruder printer and repeated all your steps, Iokanaft. What are those steps exactly that you took to reproduce this?

lokanaft, 2017-07-24:
Just create printer and save profile. All settings for extruders in profile merged after save.

Ghostkeeper, 2017-08-07:
I can't do that:

* Create printer: Ultimaker 3.
* Save profile... Option is disabled because there are no changed settings yet.

Second attempt:

* Create printer: Ultimaker 3.
* Change Printing Temperature to 201 on Extruder 1.
* Click on profile drop-down, select "Create profile from current settings/overrides".
* Name the profile: "Test". Close profile manager.
* Printing Temperature is still 201 on Extruder 1.
* Switch to Extruder 2. Printing temperature is 200 there.

It displays the wrong temperature in the printer manager though. That seems to be the bug I just arrived at in https://github.com/Ultimaker/Cura/issues/2151 where it doesn't update settings in the table that aren't listed in the selected profile.

closed 2018-02-21 by ianpaschal