Ultimaker/Cura #1808: Errors when adding printers

fieldOfView, 2017-05-08:
Since the recent merge of the ContainerStack work, I'm running into lots of errors when adding (or subsequently working with) multiextrusion printers (including the UM3).

Steps to reproduce:
* clear Cura config
* start Cura
* add UM3

```
2017-05-08 10:03:44,004 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-08 10:03:44,007 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-08 10:03:44,008 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Cura\cura\Settings\MachineManager.py", line 299, in _updateStacksHaveErrors
2017-05-08 10:03:44,009 - CRITICAL - cura.CrashHandler.show [39]:     self._stacks_have_errors = self._checkStacksHaveErrors()
2017-05-08 10:03:44,009 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Cura\cura\Settings\MachineManager.py", line 366, in _checkStacksHaveErrors
2017-05-08 10:03:44,010 - CRITICAL - cura.CrashHandler.show [39]:     if stack.hasErrors():
2017-05-08 10:03:44,011 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\ContainerStack.py", line 581, in hasErrors
2017-05-08 10:03:44,012 - CRITICAL - cura.CrashHandler.show [39]:     enabled = self.getProperty(key, "enabled")
2017-05-08 10:03:44,013 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Cura\cura\Settings\ExtruderStack.py", line 62, in getProperty
2017-05-08 10:03:44,014 - CRITICAL - cura.CrashHandler.show [39]:     return super().getProperty(key, property_name)
2017-05-08 10:03:44,015 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\ContainerStack.py", line 199, in getProperty
2017-05-08 10:03:44,016 - CRITICAL - cura.CrashHandler.show [39]:     return value(self)
2017-05-08 10:03:44,016 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingFunction.py", line 65, in __call__
2017-05-08 10:03:44,017 - CRITICAL - cura.CrashHandler.show [39]:     for name in self._used_values:
2017-05-08 10:03:44,018 - CRITICAL - cura.CrashHandler.show [39]: AttributeError: 'SettingFunction' object has no attribute '_used_values'
```

Interestingly, sometimes this is logged as a WARNING, other times it is logged as CRITICAL.

nallath, 2017-05-08:
The only way that this can go wrong if that the setting function is called before it's constructed. Which seems rather impossible to me. @awhiemstra Any idea what could cause this?

fieldOfView, 2017-05-08:
The issue went away, magically. I can no longer reproduce it, though I could from friday morning to this morning. Issues that magically go away are scary, because they could magically reappear. This has race-conditional all over it.

awhiemstra, 2017-05-09:
The problem is with the cached definitions. They include the SettingFunction which received some code changes. The reason things "magically went away" is probably because there was a change in fdmprinter that triggered a rebuild of the cached definitions.

This should not happen for the final release, since we explicitly make sure to use a clean cache when we start a new version for the first time.

closed 2017-05-09 by awhiemstra

fieldOfView, 2017-05-09:
Ah, yes, and since cache is no longer in the same folder as the config, I did not properly clear it when clearing out my config folder. Good analysis!