Ultimaker/Cura #1810: Bad print order - Cura prints on top of filament it just deposited

dinofx, 2017-05-09:
When layers contain a bunch of different "islands", like in the retraction test shape (http://www.thingiverse.com/thing:607968), Cura will order the islands such that the first island printed in a layer was often the last island printed in the preceding layer.

Typically, you want hot filament to have at least 10 seconds to cool before you try printing on top of it.  Otherwise, the new filament being laid down will pull, squish or distort the wet layer trying to support it.

There needs to be a way to make Cura visit "islands" in the roughly the same order, so that the filament has a chance to cool before more filament is laid down on top.

BagelOrb, 2017-05-09:
Which Cura version are you using?

This should be addressed in Cura 2.5

dinofx, 2017-05-09:
I downloaded 2.5, but it would launch on my Mac.  I was using 2.4.

BagelOrb, 2017-05-10:
You mean that 2.5 wouldn't launch??!

Could you please post the logs?

dinofx, 2017-05-10:
It wouldn't launch on one of my MacBooks.  I tried a different one and it is working.  This issue seems to be resolved in 2.5.  Where would the log be located?  If I can find something, I'll open a separate issue.

closed 2017-05-10 by dinofx

Ghostkeeper, 2017-05-12:
For posterity, 2.4 had this as a setting: "Start Layers At Same Position". Instead of minimising the number of retractions it had to make (reducing scars and print time) it would always start with the same piece so that the layer has more time to cool. In 2.5 this setting is always on, because multi-threading this part per layer prevents us from knowing where the previous layer ends since the previous layer may not yet have been computed.

To find the log, see https://github.com/Ultimaker/Cura#logging-issues