Ultimaker/Cura #1812: ImportError: No module named 'UM.FlameProfiler'

tylerecouture, 2017-05-10:
I recently updated to 2.5 (on Ubuntu) and I'm getting this error:

When I try to run it:
```
cura.CrashHandler.show [36]: An uncaught exception has occurred!

cura.CrashHandler.show [39]: Traceback (most recent call last):

cura.CrashHandler.show [39]:   File "/usr/bin/cura", line 54, in

cura.CrashHandler.show [39]:     import cura.CuraApplication

cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 42, in

cura.CrashHandler.show [39]:     from cura.Settings.ProfilesModel import ProfilesModel

cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/cura/Settings/ProfilesModel.py", line 10, in

cura.CrashHandler.show [39]:     from cura.QualityManager import QualityManager

cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/cura/QualityManager.py", line 12, in

cura.CrashHandler.show [39]:     from cura.Settings.ExtruderManager import ExtruderManager

cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/cura/Settings/ExtruderManager.py", line 5, in

cura.CrashHandler.show [39]:     from UM.FlameProfiler import pyqtSlot

cura.CrashHandler.show [39]: ImportError: No module named 'UM.FlameProfiler'
```

fieldOfView, 2017-05-12:
How are you running Cura? From source? With the PPA? With the AppImage?

If you are running Cura from source, make sure you also pull in the 2.5 branch of Uranium.

tylerecouture, 2017-05-12:
From the ppa

Ghostkeeper, 2017-05-16:
I think @thopiekar 's PPA is not syncing Uranium well then.

closed 2017-07-27 by nallath