Ultimaker/Cura #1815: Cura OSX 2.5.0 crashing on start

AletheianAlex, 2017-05-11:
The problem seems to be with the included definitions.  I seem to have sidestepped the problem by viewing the contents of the Cura.app package and deleting absolutely everything in the definitions, extruders, materials, meshes, and variants folders (as well as clearing out everything unnecessary in the application support folder), and then dragging just the definitions that I need into the respective folders in the Cura app package.  It runs just fine then, and boots up in a fraction of the time.

Here is the apple crash report and Cura log for reference:

++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
OS X crash report
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Process:               cura [98091]
Path:                  /Applications/Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               2.5 (2.5.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [98091]
User ID:               501

Date/Time:             2017-05-10 21:35:46.229 -0400
OS Version:            Mac OS X 10.12.4 (16E195)

System Integrity Protection: enabled

Crashed Thread:        34  QQmlThread

Exception Type:        EXC_BAD_ACCESS (SIGBUS)
Exception Codes:       KERN_PROTECTION_FAILURE at 0x000000061f19d720
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Bus error: 10
Termination Reason:    Namespace SIGNAL, Code 0xa
Terminating Process:   exc handler [0]

VM Regions Near 0x61f19d720:
    MALLOC_LARGE           000000061eeac000-000000061f12e000 [ 2568K] rw-/rwx SM=PRV  
--> JS JIT generated code  000000061f198000-000000061f19e000 [   24K] r-x/rwx SM=PRV  
    MALLOC_LARGE           000000061f1a5000-000000061f2e6000 [ 1284K] rw-/rwx SM=PRV  

Thread 34 Crashed:: QQmlThread
0   libsystem_platform.dylib      	0x00007fffaa42206a _platform_memmove$VARIANT$Nehalem + 138
1   libQt5Qml.5.dylib             	0x0000000110037e30 JSC::AssemblerBuffer::executableCopy(JSC::JSGlobalData&, void*, JSC::JITCompilationEffort) + 160
2   libQt5Qml.5.dylib             	0x0000000110037d27 JSC::LinkBuffer::linkCode(void*, JSC::JITCompilationEffort) + 39
3   libQt5Qml.5.dylib             	0x00000001100265ec QV4::JIT::Assembler::link(int*) + 1068
4   libQt5Qml.5.dylib             	0x0000000110028519 QV4::JIT::InstructionSelection::run(int) + 2425
5   libQt5Qml.5.dylib             	0x000000010ffb976c QV4::EvalInstructionSelection::compile(bool) + 60
6   libQt5Qml.5.dylib             	0x000000010fff1417 QQmlTypeCompiler::compile() + 4407
7   libQt5Qml.5.dylib             	0x000000011014a3c1 QQmlTypeData::compile() + 97
8   libQt5Qml.5.dylib             	0x0000000110149fd4 QQmlTypeData::done() + 2212
9   libQt5Qml.5.dylib             	0x0000000110142e1b QQmlDataBlob::tryDone() + 91
10  libQt5Qml.5.dylib             	0x00000001101447d4 QQmlTypeLoader::setData(QQmlDataBlob*, QQmlDataBlob::Data const&) + 212
11  libQt5Qml.5.dylib             	0x0000000110144095 QQmlTypeLoader::loadThread(QQmlDataBlob*) + 885
12  libQt5Qml.5.dylib             	0x0000000110143802 QQmlTypeLoaderThread::loadThread(QQmlDataBlob*) + 18
13  libQt5Qml.5.dylib             	0x00000001101a8c31 QQmlThreadPrivate::threadEvent() + 97
14  libQt5Qml.5.dylib             	0x00000001101a8bc0 QQmlThreadPrivate::event(QEvent*) + 32
15  libQt5Widgets.5.dylib         	0x00000001109bba7a QApplicationPrivate::notify_helper(QObject*, QEvent*) + 314
16  libQt5Widgets.5.dylib         	0x00000001109bceae QApplication::notify(QObject*, QEvent*) + 510
17  PyQt5.QtWidgets.so            	0x0000000110417662 sipQApplication::notify(QObject*, QEvent*) + 114
18  libQt5Core.5.dylib            	0x000000010f148af4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 164
19  libQt5Core.5.dylib            	0x000000010f14985b QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 891
20  libQt5Core.5.dylib            	0x000000010f19ddd9 QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 73
21  libQt5Core.5.dylib            	0x000000010f144ce1 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 417
22  libQt5Core.5.dylib            	0x000000010ef7df43 QThread::exec() + 115
23  libQt5Qml.5.dylib             	0x00000001101a8cc9 QQmlThreadPrivate::run() + 57
24  libQt5Core.5.dylib            	0x000000010ef818fa QThreadPrivate::start(void*) + 362
25  libsystem_pthread.dylib       	0x00007fffaa42c9af _pthread_body + 180
26  libsystem_pthread.dylib       	0x00007fffaa42c8fb _pthread_start + 286
27  libsystem_pthread.dylib       	0x00007fffaa42c101 thread_start + 13

Thread 34 crashed with X86 Thread State (64-bit):
  rax: 0x000000061f19d720  rbx: 0x0000000000000814  rcx: 0x0000000000000010  rdx: 0x00000000000000e4
  rdi: 0x000000061f19d730  rsi: 0x00007fe27fb0b480  rbp: 0x0000700008467f90  rsp: 0x0000700008467f90
[... 148 lines cut ...]
2017-05-10 21:35:41,891 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:41,936 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:42,069 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:42,114 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:42,248 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:42,294 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:42,431 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:42,479 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:42,618 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:42,667 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:42,702 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:42,747 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:42,914 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:42,963 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:43,124 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-10 21:35:43,174 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-10 21:35:43,244 - CRITICAL - UM.Settings.ContainerRegistry.load [181]: Found a container with a duplicate ID: M200_settings
2017-05-10 21:35:43,246 - CRITICAL - UM.Settings.ContainerRegistry.load [182]: Existing container is <UM.Settings.InstanceContainer.InstanceContainer object at 0x628005677118>, trying to load <class 'UM.Settings.ContainerStack.ContainerStack'> from /Users/user/Library/Application Support/cura/machine_instances/M200_settings.stack.cfg
2017-05-10 21:35:43,247 - DEBUG - UM.Settings.ContainerRegistry.load [204]: Loading data into container registry took 10.70019793510437 seconds
2017-05-10 21:35:43,254 - DEBUG - UM.Controller.setActiveView [81]: Setting active view to SolidView
2017-05-10 21:35:43,260 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine ACK_MP, which doesn't exist.
[... cut ...] r trains for machine ACK_MP, which doesn't exist.
2017-05-10 21:35:44,622 - DEBUG - UM.Qt.Bindings.Theme.load [126]: Loading theme file: /Applications/Cura.app/Contents/MacOS/resources/themes/cura/theme.json
2017-05-10 21:35:44,725 - DEBUG - UM.Qt.Bindings.Theme.load [186]: Loaded theme /Applications/Cura.app/Contents/MacOS/resources/themes/cura
2017-05-10 21:35:45,545 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [134]: Creating additional ui components for UM3.

awhiemstra, 2017-05-11:
The important bits here are:

> 2017-05-10 21:35:43,244 - CRITICAL - UM.Settings.ContainerRegistry.load [181]: Found a container with a duplicate ID: M200_settings
> 2017-05-10 21:35:43,246 - CRITICAL - [...]

This suggests a corrupted local configuration. Please try either deleting the /Users/user/Library/Application Support/cura directory (if you have no important data) or renaming it/moving it somewhere else.

Note that you should not modify the application bundle directly, there is a lot of important data there that is needed for Cura. Additionally, when OSX is set to only run signed applications, it will refuse to run a modified bundle.

AletheianAlex, 2017-05-11:
The M200 profile was irrelevant: I tried importing that one the other day after I tried a clean re-install to test the import function since that has not been working, but the crashing problem was there before as well as after clean re-installs.  I should have removed that profile before starting up the app to capture the crash log so as not to be a red herring.  Removing the definitions in the app package was just a kludge, but it seems to work for now.

closed 2018-01-22 by ChrisTerBeke