Ultimaker/Cura #1818: IMPORT materiaux

ERIC-3D, 2017-05-11:
Bonjour,
J'ai fais plusieurs fois importer le fichier PETG , ce dernier s'importe avec succès mais n’apparaît pas dans la liste des matériaux !
Pouvez vous m'aidez à résoudre le problème ? 
Cordialement MR ERIC.

fieldOfView, 2017-05-12:
The language spoken in the Cura github project is english. Also note that the issue-queue is not a support forum. Perhaps you can use the Ultimaker forums, they have a section for "other languages" too:
https://ultimaker.com/en/community/languages/other-languages

The people at that forum will need more information to help you: What version of Cura are you using, what printer, what OS?

Ghostkeeper, 2017-05-12:
The problem is probably that the PETG material that's being imported has a different diameter than the printer requires, or specifically says it doesn't support the printer.

fieldOfView, 2017-05-12:
>  that's being imported has a different diameter than the printer requires

That would only be the case if he's running master, since Cura 2.5 does no such filtering.

BagelOrb, 2017-07-23:
I understand it's fixed now.

closed 2017-07-23 by BagelOrb