Ultimaker/Cura #1820: Unable to restore default settings once profile saved

tcm0116, 2017-05-12:
If I edit a setting in a profile and then save the profile, there doesn't appear to be any way to remove that override from the profile other than by manually editing the .cfg file externally to Cura.

Ghostkeeper, 2017-05-12:
Indeed there is none. Should there be? I think the use case for this feature is very small.

You can set it back in the interface to whatever value the original profile had, but not if the original profile scaled in combination with other settings.

tcm0116, 2017-05-12:
The most significant case is if the print temperature is changed and then the quality profile is saved. When that happens, the temperature defined I'm the selected material profile will never be utilized unless that setting is manually removed from the quality profile. 

Another case is for dynamically computed values. Once a manual value is entered and saved, there is no way to go back to using the dynamically computed value without manually editing the profile.

nallath, 2017-05-12:
It should show an "i" icon next to the setting if its dynamicly calculated. Pressing that should restore the calculated value.

tcm0116, 2017-05-12:
I didn't realize that. Thanks for pointing​ it out. 

I still think it would be nice to have a similar option for non-cannulated values.

fieldOfView, 2017-05-12:
A long, long time ago, I drew up this:

[image]

The ability to add/remove settings from a profile, and even edit the values of settings in profiles.

nallath, 2017-05-12:
I don't think i ever saw that design. Shouldn't be that hard to implement. Especially not if you only want to  remove  a setting from a profile.

tcm0116, 2017-05-12:
@fieldOfView  I really like that!

fvrmr, 2021-07-27:
We have a ticket on our backlog for this. Developers see CURA-6722

GregValiant, 2024-11-13:
@tcm0116 does the FieldOfView "Material Settings" plugin address this request bug?

closed 2024-12-10 by github-actions[bot]