Ultimaker/Cura #1822: Cura 2.5.0 crash on start (Mac)

nicolinux, 2017-05-12:
All of the sudden Cura 2.5 won't start anymore. I have deleted the contents of cura.log and started it again:

```
2017-05-12 12:43:33,409 - DEBUG - UM.Backend.Backend._logSocketState [172]: Socket state changed to Connected
2017-05-12 12:43:33,411 - DEBUG - UM.Backend.Backend._onSocketStateChanged [162]: Backend connected on port 49674
2017-05-12 12:43:33,641 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [72]: Not attaching MachineSettingsAction to ultimaker_original_dual; Multi-extrusion printers are not supported
2017-05-12 12:43:33,884 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [72]: Not attaching MachineSettingsAction to vertex_k8400_dual; Multi-extrusion printers are not supported
2017-05-12 12:43:34,296 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [72]: Not attaching MachineSettingsAction to ord; Multi-extrusion printers are not supported
2017-05-12 12:43:34,693 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [72]: Not attaching MachineSettingsAction to ultimaker3_extended; Multi-extrusion printers are not supported
2017-05-12 12:43:35,308 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [72]: Not attaching MachineSettingsAction to punchtec_connect_xl; Multi-extrusion printers are not supported
2017-05-12 12:43:35,362 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [72]: Not attaching MachineSettingsAction to ultimaker3; Multi-extrusion printers are not supported
2017-05-12 12:43:35,459 - DEBUG - MachineSettingsAction.MachineSettingsAction._onContainerAdded [72]: Not attaching MachineSettingsAction to cartesio; Multi-extrusion printers are not supported
2017-05-12 12:43:38,039 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [457]: Unsupported material setting retraction prime speed
2017-05-12 12:43:38,040 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [457]: Unsupported material setting nozzle switch prime speed
2017-05-12 12:43:38,126 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
[...]
2017-05-12 12:43:38,260 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [457]: Unsupported material setting retraction prime speed
2017-05-12 12:43:38,260 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [457]: Unsupported material setting nozzle switch prime speed
2017-05-12 12:43:38,395 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-12 12:43:38,413 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
[...]
2017-05-12 12:43:39,059 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [457]: Unsupported material setting retraction prime speed
2017-05-12 12:43:39,059 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [457]: Unsupported material setting nozzle switch prime speed
2017-05-12 12:43:39,139 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3
2017-05-12 12:43:39,161 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
[...]
2017-05-12 12:43:44,294 - DEBUG - UM.Settings.ContainerRegistry.load [204]: Loading data into container registry took 10.800412893295288 seconds
2017-05-12 12:43:44,300 - DEBUG - UM.Controller.setActiveView [81]: Setting active view to SolidView
2017-05-12 12:43:44,307 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Ultimaker 2+, which doesn't exist.
2017-05-12 12:43:44,309 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Ultimaker 2+, which doesn't exist.
2017-05-12 12:43:45,332 - DEBUG - UM.Qt.Bindings.Theme.load [126]: Loading theme file: /Applications/Cura.app/Contents/MacOS/resources/themes/cura/theme.json
2017-05-12 12:43:45,457 - DEBUG - UM.Qt.Bindings.Theme.load [186]: Loaded theme /Applications/Cura.app/Contents/MacOS/resources/themes/cura
2017-05-12 12:43:46,368 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [134]: Creating additional ui components for UM3.
2017-05-12 12:43:46,621 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [376]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.010016202926635742 seconds. 265 input vertices. 265 output vertices.
```

nicolinux, 2017-05-12:
If I move away the contents of `~/Library/Application Support/Cura/user/`, then I can start Cura but it has no printers defined. If I move the .cfg files back, it won't start.
If I keep trying to start Cura, then at some point it will work.

rekom, 2017-05-13:
I had the same problem. Unistalled Cura, deleted Macintosh HD\Users[user_name]\Library\Application Support\cura, resintalled my config without copying the .cfg and it works again.

Ghostkeeper, 2017-05-15:
In the log, Cura seems to start fine. It starts the back-end, loads some materials, loads the theme and creates the window, then even seems to load a model and compute the collision area for it.

There are two things that stand out to me. The unsupported material settings `retraction prime speed` and `nozzle switch prime speed`. I think I remember those were indeed present in some materials in the 2.5 release but weren't a problem because the unsupported settings are just skipped.

Further, it says that it tried to get the extruder trains for UM2+. I'd expect it to not show that machine as added then, and perhaps display the add-printer wizard. You say that it just didn't start at all?

closed 2018-02-21 by ianpaschal