Ultimaker/Cura #1823: USB connected Ultimaker 2+ shows unusable "Print via USB" button

michaelrsweet, 2017-05-12:
Cura 2 shows a "Print via USB" button when an Ultimaker 2+ is connected via USB, but if I click on it I am told I can't do that.

Please either fix Cura to support USB printing to this printer or disable/hide the "Print via USB" button when it isn't supported.

fieldOfView, 2017-05-12:
If you set the gcode flavor to Reprap, usb printing will work fine

michaelrsweet, 2017-05-12:
Indeed, that does get me printing, but that brings us back to the original issue: seems like the software should be smart enough to either a) switch UltiCode to Reprap/Marlin automatically since it is compatible or b) not show the print via USB button at all. I mean, as soon as I connected the printer it showed the USB print button, overriding the previous save button selection...

(the Cura UI already supports alert balloons, so even for option A you could show a "using Reprap mode for this printer" alert or something if you are concerned about doing something without informing the user...)

nallath, 2017-05-12:
It should, but as is with any software project; there are bigger fish to fry.

Ghostkeeper, 2017-08-20:
Our project manager just rejected this issue because it was more than 3 months old.

Ghostkeeper, 2017-08-29:
Yes, he tends to do that on Sunday quite often! I've heard that some people handle administrative stuff outside of work hours, like at 2 a.m. and such. Weirdoes. I'll never understand them.

fieldOfView, 2017-08-30:
Offtopic: Kidding aside, I appreciate anyone putting in work off the clock. Tip of the hat.

closed 2023-06-13 by github-actions[bot]