Ultimaker/Cura #1827: Install Instructions not available for Ubuntu 14.04

chintanjosh, 2017-05-15:
I have been searching the net for days but haven't found anything that instructs on how to install this.

Although I have the application's image and can the run the application through the image, what I want to do it is run it through the code, so that I can know what is going on.

I am trying it for 2.5 version. 

There are no clear instruction anywhere, can anyone please give some step by step approach on how to install this ?

nallath, 2017-05-15:
Running the code is reasonably easy; just run cura_app.py through python3 and resolve the import issues that you get. I don't have a full list of packages that we depend on on hand right now.

chintanjosh, 2017-05-15:
@nallath Thanks, will do that and will update the steps I follow, for now I am facing UM import issue, Will update how things proceed further

chintanjosh, 2017-05-15:
Nope, the list goes on and on..
First it was UM, I cloned the Uranium repo and moved the UM folder in Cura folder, then it was Arcus import error,
I removed that as it was unused import
Then it is PyQt5.QtQml import error

this keeps on going.. 
Need some step by step approach

fieldOfView, 2017-05-15:
Cura is a big project, and running it from source is not as easy as checking out one or two repositories.

You need these repositories:
* Cura
* Uranium
* CuraEngine
* libArcus
* libSavitar

In order to build CuraEngine, you first need to build libArcus. In order to build libArcus, you need Google Protobuf.

You also need Numpy, Scipy, and a couple of other Python modules.

NB: do not put Uranium inside the Cura directory. Instead set PYTHON_PATH to where you have cloned Uranium.

chintanjosh, 2017-05-15:
Okay, So I need following repos:

1. Cura : https://github.com/Ultimaker/Cura/tree/2.5
2. cura-build: https://github.com/Ultimaker/cura-build
3. CuraEngine: https://github.com/Ultimaker/CuraEngine
4. Arcus: https://github.com/Ultimaker/libArcus
5. Uranium: https://github.com/Ultimaker/Uranium
6. Protobuf: https://github.com/google/protobuf/releases

And each one is are to be kept in different folder,
Now, I deleted all the things I had previously, I have made a single folder named
`cura-folder` within which I have these repos.

@fieldOfView As you said, CuraEngine>libArcus>Protobuf
So shall I start with protobuf first ?

fieldOfView, 2017-05-15:
Yes, that's how I got it to work.
But it may also be that cura-build works out of the box for you. Check its readme.

chintanjosh, 2017-05-15:
Okay thanks will look into that :slightly_smiling_face:

chintanjosh, 2017-05-17:
I have tried to use this 
https://code.alephobjects.com/source/Cura2build/

and do the steps from there but alas :disappointed: 

isn't there any easier way to do this.. 
these dependencies and their respective errors are nesting..

fieldOfView, 2017-05-17:
"alas" does not give us much to go on trying to help you. Where do you get stuck?

The easier way to do this is cura-build.

Note that AlephObjects are working on their own fork. If you want help building their fork, please ask them.

fieldOfView, 2017-05-17:
On Windows I use a hybrid approach (source/binary), where I only use the Cura and Uranium repos. I use binaries from the official releases for CuraEngine and libArcus (and a protobuf wheel I found somewhere).

cura-folder
  Cura
    [repository files...]
  Uranium
    [repository files...]
  bin
    curaengine.exe
    libArcus.pyd

I set PYTHONPATH to include both the Uranium and bin folders.

The same approach could work on linux; get curaengine and libarcus.so (?) from an distributed version of cura.

chintanjosh, 2017-05-17:
Sorry for that, 
`make[2]: *** [CMakeFiles/_CuraEngine.dir/src/commandSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/_CuraEngine.dir/all] Error 2
make: *** [all] Error 2`

the issue is with the CuraEngine's build this is while doing it from the AlephObjects Cura2build...

While going for cura-build I face this

`ImportError: No module named 'PyQt5'
CMake Error at /home/xyz/Downloads/cmake-3.7.2-Linux-x86_64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find PyQt: Found unsuitable version "", but required is at least
  "5.6" (found )
`
I am not sure if I have PyQt5 in the correct manner, but I checked it with 
python3

and doing `import PyQt5` it works, so I am not getting how to point it here

chintanjosh, 2017-05-17:
I tried this
Downloaded PyQt5 and extracted it and then
export PyQt_PATH=/path/to/PyQt5_gpl-5.8.2

then removed build folder 
then 
`mkdir build && cd build`
and then 
`cmake ..` 
but the same error `CMake Error at /home/xyz/Downloads/cmake-3.7.2-Linux-x86_64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find PyQt: Found unsuitable version "", but required is at least
  "5.6" (found )
`

Ghostkeeper, 2017-05-17:
It could be finding an older version from your Ubuntu installation instead, and be unable to find the version number of that version?

chintanjosh, 2017-05-17:
I only have single version of PyQt5 installed, 
besides it is not finding any version maybe, because its not there in error message,
`Could NOT find PyQt: Found unsuitable version "", but required is at least "5.6" (found )`
 I don't know how to point that to cmake that I do have that version

awhiemstra, 2017-05-17:
It sounds like you downloaded the source code for PyQt. This will still need to be built, which needs a recent Qt. As far as I know, 14.04 has something like 5.5 or so, which is too old at this point.

chintanjosh, 2017-05-17:
For Qt version check I ran 
`qmake -version`

`QMake version 3.0
Using Qt version 5.6.1 in /home/xyz/Qt5.6.1/5.6/gcc_64/lib
`

Then I went to downloaded source of PyQt and have executed these steps 
`python3 configure.py --qmake=/usr/bin/qmake`
`make`

That step is still going on. Lets see what happens

chintanjosh, 2017-05-17:
Nope, still the same, is there any PATH variable that I need to set ?

chintanjosh, 2017-05-17:
@fieldOfView As you said , I directly tried to run cura_app.py
It said 
`Segmentaion Fault (Core Dumped)`

Prior to it there was error regarding sip
Saying that for PyQt5.QtQml to work we need api version 12.1 but we have 11.0

so I updated the sip version
and after that this segmentaion error came.

Any idea what it could be ?

fieldOfView, 2017-05-17:
Segmentation faults are well beyond my comfort zone. Sorry, can't help you there.

chintanjosh, 2017-05-17:
@fieldOfView Thanks anyways :)
Any idea on how to set this PyQt path for cura-build ?

chintanjosh, 2017-05-18:
@fieldOfView Hi, I went ahead and solved that segmentation fault, seems like they needed a different version of python, so I went ahead and did everything with python3.5

After that faced other few errors and gradually solved them..
This one however is something that I am not getting. 
Any ideas ?
Below is the attached file of log
this comes when I run 
`python3.5 cura_app.py`
[error_log.txt](https://github.com/Ultimaker/Cura/files/1010032/error_log.txt)

fieldOfView, 2017-05-18:
You need to clone or link the https://github.com/ultimaker/fdm_materials repository into Cura/resources/materials
I don't know if that would solve the infinite recursion, but it might.

chintanjosh, 2017-05-18:
@fieldOfView Yes, thanks It did solved the issue of infinite recursion.

Next was this 
`17-05-18 12:54:47,119 - DEBUG - UM.Qt.Bindings.Theme.load [186]: Loaded theme /home/chintan/Desktop/cura-folder/Cura/cura/../resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed

(python3.5:20974): Gtk-CRITICAL **: IA__gtk_widget_style_get: assertion 'GTK_IS_WIDGET (widget)' failed
`

fieldOfView, 2017-05-18:
That suggests to me that something is still not quite right with your Qt or pyqt installation.
Have you tried a simple pyqt "hello world" application? Eg http://pythonprogramminglanguage.com/pyqt5-hello-world/

chintanjosh, 2017-05-18:
It was a glitch maybe, I re-ran it and everything goes smooth now.

For any one stumbling on this in future.

Get all the dependencies first, then go for the kill.

All the dependencies are listed on Cura's page and cura-build's page.

Thank you @all for the support.

closed 2017-05-18 by chintanjosh