Ultimaker/Cura #1828: High quality support structure

Jip-Hop, 2017-05-15:
I don't want to remove the support structure, but leave the supports attached. In my case they should become part of the thing. So it's desirable to treat the generated support structures in a similar way as the rest of the model.

One of the issues I currently face is stringing between the supports, as mentioned in issue #724 .

If high quality supports would be available I could also start to experiment with printing transparent/clear support with [XT_CLEAR](http://colorfabb.com/xt).

Ghostkeeper, 2017-05-16:
There's an experimental setting called "Make Overhang Printable", with an accompanying Maximum Model Angle. If you enable Make Overhang Printable and then set the Maximum Model Angle to 0 then you get something that might get close to what you wanted.

Jip-Hop, 2017-05-16:
Yes it's almost what I want! Except I need to print the support structure in a different material than the model. This option will add parts below the overhang instead of support structure, but will use the same material as the rest of the model unfortunately. But thanks for the hint. Didn't know about that feature.

Jip-Hop, 2017-05-16:
"Make Overhang Printable" is a very cool feature but it doesn't eliminate the need for support completely.
[image]
So my request for high quality supports remains.

Ghostkeeper, 2017-05-17:
Yeah, it's only useful for eliminating small corners that would otherwise create tiny pillars of support that would fall over anyway. That's why it's in Experimental, really.

If the "support" needs to be printed with a second extruder the Make Overhang Printable solution won't work for you, then I see no other solution than actually modelling the support structures.

You can fairly easily do that in OpenSCAD. I still had a linear_extrude function laying around. I adapted it for your purpose:

    max_height = 1000; //Some very big number, higher than the maximum height of your model.
    max_width = 1000; //Some very big number, greater than the maximum X or Y coordinate of any vertex.

    difference() {
        translate([0, 0, -max_height]) {
            difference() {
                minkowski() {
                    import("your_model.stl");
                    scale([0, 0, max_height])
                        cube(1);
                }
                translate([-max_width, -max_width, 0])
                    cube([max_width * 2, max_width * 2, max_height]);
            }
        }
        import("your_model.stl");
    }

That should generate a model that exactly supports all overhang in your print. You can merge that with your normal model in Cura and set the support model to be printed with the support extruder.

Jip-Hop, 2017-05-17:
Great suggestion to use OpenSCAD. Haven't used it before but I think suits my project perfectly (especially since the boolean operations work much more reliable compared to Blender concluding from my tests today).

However I can't get your code to work. I get the following error:

`Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: pe_prev->is_border() || !internal::Plane_constructor::get_plane(pe_prev->facet(),pe_prev->facet()->plane()).is_degenerate() File: /Users/kintel/code/OpenSCAD/openscad-2015.03-3/../libraries/install/include/CGAL/Nef_3/polyhedron_3_to_nef_3.h Line: 251 
Geometries in cache: 11862
Geometry cache size in bytes: 20885584
CGAL Polyhedrons in cache: 88
CGAL cache size in bytes: 104750456
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.`

Ghostkeeper, 2017-05-18:
It says your model is degenerate. Being "degenerate" could have a multitude of meanings, like having multiple vertices on top of each other, not being watertight, etc.

Perhaps you should look into mesh fixing programs first. Does X-Ray View in Cura show anything red?

Jip-Hop, 2017-05-18:
Tried it with this STL. Couldn't be more simple. I'm quite sure it's watertight.
[simple_stl.stl.zip](https://github.com/Ultimaker/Cura/files/1010622/simple_stl.stl.zip)

It's a good suggestion to model supports manually, but I'd still like a setting in Cura to prevent supports stringing (by enabling retraction for supports).

Ghostkeeper, 2017-05-18:
Let that be the feature request then: Retractions on support.

Jip-Hop, 2017-05-22:
@Ghostkeeper your "Make Overhang Printable" tip inspired me. I've been looking into OpenSCAD ever since and can confirm your linear_extrude function is working. STL importing into OpenSCAD is a bit tricky, that's why it wasn't working for me before.

Do you know of a way to automatically generate something similar to "Make Overhang Printable" and  your "linear_extrude" in OpenSCAD?

If I can generate the green parts in this picture with OpenSCAD (with a specifiable angle and min support thickness), then my prints with transparent support would be even better.
[image]

"Make Overhang Printable" limits the amount of support, so that's great. But it's using the same material as the model itself (I want to print those generated parts in a transparent material, same as the support material).

Also, it absorbs part of my model. Clearly seen here:
[image]
[image]

But if I could generate this ramp and support in OpenSCAD for a given STL then I could print them transparent and still see through the original model.

Ghostkeeper, 2017-05-24:
There's one setting "Conical Support" that basically creates the effect you want. You didn't want it as actual support, so it's not exactly what you asked for. Perhaps you can approximate printing a mesh by setting the support extruder to the extruder of that XT_CLEAR stuff, and perhaps alter some settings like the support pattern to create a border around your mesh. To simulate a top/bottom skin you could use support interface, though that won't work with the interface to the build plate so you might need to print a slab of normal model on the very first layer.

Ghostkeeper, 2017-05-24:
Alternatively you could try making a horizontal inset of the support mesh in OpenSCAD. You could try a Minkowski subtraction of a horizontal square (which is actually a cube scaled to Z=0). To do this you have to place a cube around the entire thing, subtract the support mesh, Minkowski Sum that, then subtract the result from your support mesh again. I tried that for you but didn't get it to work directly and I shouldn't spend too much time on it.

Jip-Hop, 2017-05-25:
Thanks for the useful advice, especially if this feature request (preventing stringing when printing supports) is implemented at some point.

JDunn3, 2017-07-06:
Just throwing in my 2 cents here that this is a very desirable feature. Stringing between supports is making the vast majority of my prints unprintable with Cura.

Ghostkeeper, 2017-07-09:
Actually, what is the problem with stringing between supports? Usually you won't need supports to be pretty, right? Are you worried about underextrusion?

JDunn3, 2017-07-10:
@Ghostkeeper underextrusion is definitely a problem depending on filament, especially when a small island happens to be right after a string-spot. In addition to that, with some complex(game character) models, significant stringing between supports makes it much harder to remove supports than it would otherwise be without the stringing.

bartoncasey, 2017-10-08:
I'm currently dealing with a print whose supports are causing trouble due to the lack of retraction between supports. The travel path between supports travels over a long support structure, which starts to collect ooze into fingers. Eventually the head starts banging on these fingers during its travel, which knocks the support off the bed.

Oozing over empty space would be fine.  Oozing over another support structure is bad,

fusentasticus, 2017-10-11:
With PETG oozing anywhere is bad --- the extra filament defies physics (as I understand it..) and it flows upwards on the nozzle. The ooze will end up over the model, some burnt, especially if using a nozzle silicone sleeve (to mitigate head cleaning with this gooy stuff).  So policy "always retract" is important for PETG. Thanks btw to ghostkeeper for the great discussion of using SCAD.

PvtPrivate, 2017-11-12:
Also having this problem, ooze builds up on other supports and the head eventually bangs into them, causing a missed step. This problem is very consistent as the 3 times i tried to print the model, it skipped every time.

stevenfayers, 2017-12-01:
I would also like retraction on supports.  @Ghostkeeper - the problem with stringing between supports occurs when you have parts of the real model that reside between 2 supports. The stringing then sticks to those parts of the model even if those parts of the model are slightly below the current Z position.

Ghostkeeper, 2018-03-19:
This would be implemented by a Pull Request that's currently under consideration: https://github.com/Ultimaker/CuraEngine/pull/709