Ultimaker/Cura #1829: changing extruder amount at machine settings

maukcc, 2017-05-15:
Cura Version: 2.5.99 thopiekar`s master.

When changing the extruder amount at machine settings, it will not update the amount
So in settings I can change the amount of extruders, ig from 4 to 2, but it will keep showing the 4 extruder tabs. 
Until I change the machine and then change it back again, then the value has been activated and it only shows the 2 tabs.(or visa versa)

fieldOfView, 2017-05-15:
This is a recent regression.

From the logs:
```
2017-05-15 17:00:48,244 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-15 17:00:48,254 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-15 17:00:48,256 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Cura\cura\..\plugins\MachineSettingsAction\MachineSettingsAction.py", line 192, in setMachineExtruderCount
2017-05-15 17:00:48,257 - CRITICAL - cura.CrashHandler.show [39]:     definition_changes_container.setProperty("machine_extruder_count", "value", extruder_count)
2017-05-15 17:00:48,259 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\InstanceContainer.py", line 309, in setProperty
2017-05-15 17:00:48,260 - CRITICAL - cura.CrashHandler.show [39]:     self._instances[key].setProperty(property_name, property_value, container)
2017-05-15 17:00:48,263 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 141, in setProperty
2017-05-15 17:00:48,265 - CRITICAL - cura.CrashHandler.show [39]:     self.updateRelations(container)
2017-05-15 17:00:48,266 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 219, in updateRelations
2017-05-15 17:00:48,268 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(changed_relations, self._definition.relations, property_name)
2017-05-15 17:00:48,269 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 248, in _addRelations
2017-05-15 17:00:48,271 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(relations_set, relation.target.relations, property_name)
2017-05-15 17:00:48,274 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 248, in _addRelations
2017-05-15 17:00:48,276 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(relations_set, relation.target.relations, property_name)
2017-05-15 17:00:48,277 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 248, in _addRelations
2017-05-15 17:00:48,283 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(relations_set, relation.target.relations, property_name)
2017-05-15 17:00:48,284 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 248, in _addRelations
2017-05-15 17:00:48,286 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(relations_set, relation.target.relations, property_name)
2017-05-15 17:00:48,288 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 248, in _addRelations
2017-05-15 17:00:48,292 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(relations_set, relation.target.relations, property_name)
2017-05-15 17:00:48,295 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 248, in _addRelations
2017-05-15 17:00:48,297 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(relations_set, relation.target.relations, property_name)
2017-05-15 17:00:48,299 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\Users\Aldo\Documents\Code Projects\UM\Uranium\UM\Settings\SettingInstance.py", line 248, in _addRelations
2017-05-15 17:00:48,300 - CRITICAL - cura.CrashHandler.show [39]:     self._addRelations(relations_set, relation.target.relations, property_name)
```

fieldOfView, 2017-05-15:
Last commit where it worked seems to be #4474b1684782e46891cfba527df56a783961b8e4

Ghostkeeper, 2017-05-16:
We already had this issue in our tracker; @Vandrasc found it too. See CURA-3813.

maukcc, 2017-05-17:
This is working again, so this can be closed

closed 2017-05-17 by nallath