Ultimaker/Cura #1836: Cura 2.5 freezes on startup Debian & Ubuntu

Schwarzdrossel, 2017-05-17:
Hi,
I'm currently using debian sid and Cura hangs itself on startup,
the same happens on ubuntu 17.04 and 16.04 (via Virtualbox)

last terminal output is:
-05-17 08:41:37,401 - DEBUG - UM.Backend.Backend._logSocketState [172]: Socket state changed to Connected
2017-05-17 08:41:37,408 - DEBUG - UM.Backend.Backend._onSocketStateChanged [162]: Backend connected on port 49674
2017-05-17 08:41:37,415 - DEBUG - UM.Controller.setActiveView [81]: Setting active view to SolidView

nallath, 2017-05-17:
Are you running it from PPA or appimage?

Schwarzdrossel, 2017-05-17:
Appimage

nallath, 2017-05-17:
Could you post the entire log?

Schwarzdrossel, 2017-05-17:
./Cura-2.5.0.AppImage 
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.3 Core profile
UM.Qt.QtApplication.__init__ [75]: Detected most suitable OpenGL context version: 4.1 Core profile
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 70: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 70: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 78: saw unknown, expected number
UM.PluginRegistry.loadPlugin [102]: Loaded plugin 3MFReader
UM.PluginRegistry.loadPlugin [102]: Loaded plugin 3MFWriter
UM.PluginRegistry.loadPlugin [102]: Loaded plugin AutoSave
UM.PluginRegistry.loadPlugin [102]: Loaded plugin ChangeLogPlugin
2017-05-17 13:52:45,966 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ConsoleLogger
2017-05-17 13:52:46,027 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /tmp/.mount_qOnPwU/usr/bin/CuraEngine
2017-05-17 13:52:46,031 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraEngineBackend
2017-05-17 13:52:46,051 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraProfileReader
2017-05-17 13:52:46,070 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraProfileWriter
2017-05-17 13:52:46,246 - INFO - UM.PluginRegistry.loadPlugin [85]: Plugin Doodle3D uses an incompatible API version, ignoring
2017-05-17 13:52:46,266 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin OBJReader
2017-05-17 13:52:46,282 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin OBJWriter
2017-05-17 13:52:46,292 - WARNING - STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2017-05-17 13:52:46,300 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin STLReader
2017-05-17 13:52:46,318 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin STLWriter
2017-05-17 13:52:46,329 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin FileLogger
2017-05-17 13:52:46,355 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin GCodeProfileReader
2017-05-17 13:52:46,390 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin GCodeReader
2017-05-17 13:52:46,410 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin GCodeWriter
2017-05-17 13:52:46,442 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ImageReader
2017-05-17 13:52:46,516 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LayerView
2017-05-17 13:52:46,539 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LegacyProfileReader
2017-05-17 13:52:46,567 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LocalFileOutputDevice
2017-05-17 13:52:46,594 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin MachineSettingsAction
2017-05-17 13:52:46,626 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting cura/active_mode.
2017-05-17 13:52:46,630 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin PerObjectSettingsTool
2017-05-17 13:52:46,657 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin PostProcessingPlugin
2017-05-17 13:52:46,694 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin RemovableDriveOutputDevice
2017-05-17 13:52:46,722 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SliceInfoPlugin
2017-05-17 13:52:46,745 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SolidView
2017-05-17 13:52:46,766 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CameraTool
2017-05-17 13:52:46,791 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin MirrorTool
2017-05-17 13:52:46,865 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin RotateTool
2017-05-17 13:52:46,894 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ScaleTool
2017-05-17 13:52:46,911 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SelectionTool
2017-05-17 13:52:46,935 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin TranslateTool
2017-05-17 13:52:46,967 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin UltimakerMachineActions
2017-05-17 13:52:47,322 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin UM3NetworkPrinting
2017-05-17 13:52:47,411 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin UpdateChecker
2017-05-17 13:52:47,413 - INFO - UpdateChecker.UpdateChecker.run [40]: Checking for new version of cura
2017-05-17 13:52:47,520 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin USBPrinting
2017-05-17 13:52:47,560 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin VersionUpgrade21to22
2017-05-17 13:52:47,580 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin VersionUpgrade22to24
2017-05-17 13:52:47,600 - ERROR - UM.PluginRegistry.loadPlugin [91]: Plugin VersionUpgrade24to25 did not return any objects to register
2017-05-17 13:52:47,614 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SimpleView
2017-05-17 13:52:47,634 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin WireframeView
2017-05-17 13:52:47,676 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin X3DReader
2017-05-17 13:52:47,704 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin XmlMaterialProfile
2017-05-17 13:52:47,724 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin XRayView
[... cut ...]
2017-05-17 13:53:30,704 - DEBUG - XmlMaterialProfile.XmlMaterialProfile.deserialize [506]: No variants found with ID or name AA 0.25 for machine ultimaker3_extended
2017-05-17 13:53:30,751 - DEBUG - UM.Settings.ContainerRegistry.load [204]: Loading data into container registry took 42.5812668800354 seconds
2017-05-17 13:53:30,758 - DEBUG - UM.Backend.Backend._logSocketState [172]: Socket state changed to Connected
2017-05-17 13:53:30,764 - DEBUG - UM.Backend.Backend._onSocketStateChanged [162]: Backend connected on port 49674
2017-05-17 13:53:30,769 - DEBUG - UM.Controller.setActiveView [81]: Setting active view to SolidView

nallath, 2017-05-17:
It's kind of odd; It does seem to be able to find the correct OpenGL drivers to function. 

It is rather weird that it fails to find numpy-stl, but I very much doubt that this causes the crash.

Schwarzdrossel, 2017-05-17:
I run the programm a bit longer and I get this logs

2017-05-17 15:39:25,456 - DEBUG - UM.Controller.setActiveView [81]: Setting active view to SolidView
Exception in thread zeroconf-Engine:
Traceback (most recent call last):
  File "/build/env/inst/lib/python3.5/threading.py", line 914, in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 951, in run
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 988, in handle_read
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 563, in __init__
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 611, in read_others
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 665, in read_name
IndexError: index out of range

Exception in thread zeroconf-ServiceBrowser__ultimaker._tcp.local.:
Traceback (most recent call last):
  File "/build/env/inst/lib/python3.5/threading.py", line 914, in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1164, in run
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1853, in send
OSError: [Errno 22] Invalid argument

"Keine derartige Schnittstelle »org.freedesktop.DBus.Properties« des Objekts im Pfad /org/freedesktop/NetworkManager/ActiveConnection/2"

nallath, 2017-05-18:
Could you try and remove the UM3NetworkingPlugin from the appimage? That is the one that does something with zeroconf.

closed 2018-02-21 by ianpaschal