Ultimaker/Cura #1837: Cura 2.5.99 is entirely broken

isbm, 2017-05-17:
Unit tests, anyone, before pushing to Git?

This is what happens on _any_ STL:
```
2017-05-17 08:49:42,974 - DEBUG - UM.FileHandler.ReadFileJob.run [63]: Loading file took 0.5 seconds
2017-05-17 08:49:42,981 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-17 08:49:42,982 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-17 08:49:42,982 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 677, in event
2017-05-17 08:49:42,982 - CRITICAL - cura.CrashHandler.show [39]:     return super().event(event)
2017-05-17 08:49:42,983 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 261, in event
2017-05-17 08:49:42,983 - CRITICAL - cura.CrashHandler.show [39]:     event._function_event.call()
2017-05-17 08:49:42,983 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/UM/Event.py", line 189, in call
2017-05-17 08:49:42,983 - CRITICAL - cura.CrashHandler.show [39]:     self._function(*self._args, **self._kwargs)
2017-05-17 08:49:42,984 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 314, in __performEmit
2017-05-17 08:49:42,984 - CRITICAL - cura.CrashHandler.show [39]:     func(dest, *args, **kwargs)
2017-05-17 08:49:42,984 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 1317, in _readMeshFinished
2017-05-17 08:49:42,984 - CRITICAL - cura.CrashHandler.show [39]:     if node.getBoundingBox().width < self._volume.getBoundingBox().width or node.getBoundingBox().depth < self._volume.getBoun
```

And this is what happens on parsing G-Code, that Cura 2.4.99 previous generated:

```
2017-05-17 08:49:51,915 - ERROR - UM.Logger.logException [76]: Exception: Exception occurred while loading file /home/bo/Documents/Electronics/suse-alexa/new-case/gcode/CFP_body.gcode
2017-05-17 08:49:51,915 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-05-17 08:49:51,915 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3/dist-packages/UM/FileHandler/ReadFileJob.py", line 61, in run
2017-05-17 08:49:51,915 - ERROR - UM.Logger.logException [80]:     self.setResult(self._handler.readerRead(reader, self._filename))
2017-05-17 08:49:51,916 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/python3/dist-packages/UM/Mesh/MeshFileHandler.py", line 28, in readerRead
2017-05-17 08:49:51,916 - ERROR - UM.Logger.logException [80]:     results = reader.read(file_name)
2017-05-17 08:49:51,916 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/GCodeReader/GCodeReader.py", line 354, in read
2017-05-17 08:49:51,916 - ERROR - UM.Logger.logException [80]:     machine_width = settings.getProperty("machine_width", "value")
2017-05-17 08:49:51,916 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getProperty'
```

nallath, 2017-05-17:
Uh; Using unstable master and then complaining that your settings are gone? If you want stability, use official releases (and not the master PPA)

isbm, 2017-05-17:
@nallath I don't want stability. I complain that the basic stuff isn't tested before push to the master. Maybe it is OK for you though...

nallath, 2017-05-17:
Master isn't seen as stable by us, release branches are. Tests are run, but a whole ton of things are just insanely hard to unit test (or impossible to). We're still looking into automating more tests, but it's not there yet

isbm, 2017-05-17:
@nallath ah OK. Well, at least this can be classified as a bug. :wink: Thanks anyway.

thopiekar, 2017-07-05:
I can only confirm what @nallath says. The master PPA is **NOT** meant to be used for daily usage.
Use the stable PPA or don't expect stable software, while using master.

closed 2017-07-05 by thopiekar

isbm, 2017-07-05:
@thopiekar so what is **DOES** meant to be used for daily usage on Ubuntu 16.04? :wink: This (seems yours) — https://launchpad.net/~thopiekar/+archive/ubuntu/cura — it is entirely broken on 16.04:
```
$ apt install cura
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 cura : Depends: python3-uranium but it is not going to be installed
        Recommends: python3-numpy-stl but it is not installable
E: Unable to correct problems, you have held broken packages.
```

The only way to use Cura right now is to download a static app-image and use in ugly mode, where the rest of the QT is different from the thing, microscopic fonts and have zero updates.

isbm, 2017-07-05:
BTW, @thopiekar what is this uranium thing? I cannot find much references to this package. Any hints here what repo is missing?

thopiekar, 2017-07-05:
@isbm There should be no repos or packages missing. The only issue I see is that some packages are still installed.
Uranium is a toolkit from Ultimaker. The AppImage also comes with it. I made a separate package for it, just like you normally package things for Debian/Ubuntu.
Make sure you remove all packages. Instead of disable the master repo and enable the stable repo, zou should have used ```ppa-purge```, which cares about removing all packages, which have been installed from a selected PPA.

fieldOfView, 2017-07-05:
Uranium in this context is https://github.com/Ultimaker/Uranium