Ultimaker/Cura #1840: Conflicts between printer name/id and definition ids

fieldOfView, 2017-05-17:
When adding a new printer with the exact id of a machine definition, the printer can not be succesfully deserialised when restarting Cura.

This is probably a regression introduced by the CuraContainerStack work.

Steps to reproduce:
* add a printer named "ultimaker3" (this exact spelling)
* close and restart Cura

Result:
The added "ultimaker3" printer is no longer listed. 
From the logs:
```
2017-05-17 13:39:26,655 - CRITICAL - UM.Settings.ContainerRegistry.load [175]: Found a container with a duplicate ID: ultimaker3
2017-05-17 13:39:26,657 - CRITICAL - UM.Settings.ContainerRegistry.load [176]: Existing container is <UM.Settings.DefinitionContainer.DefinitionContainer object at 0x0000026BBDFADA68>, trying to load <class 'cura.Settings.GlobalStack.GlobalStack'> from C:\Users\Aldo\AppData\Roaming\cura\master\machine_instances\ultimaker3.global.cfg
```

This became apparent when I added a 101Hero printer, which has the same name and id (whereas most other definitions have a different spelling between name and id, making this issue less apparent).

nallath, 2017-05-17:
I just tested this and it was already an issue with 2.5
I don't think we ever checked for globally unique names...

fieldOfView, 2017-05-17:
Shows you how many users use the 101Hero definition ;-)

> I don't think we ever checked for globally unique names
We really should.

Ghostkeeper, 2017-05-18:
Could reproduce this. It seems to be case sensitive (at least on Ubuntu).

fieldOfView, 2017-05-18:
Yes, it is case-sensitive on Windows too.

fieldOfView, 2017-05-18:
Same issue when naming a printer after a quality profile id (eg "draft").

closed 2017-05-24 by nallath