Ultimaker/Cura #1841: Print freezes on big prints

partsoven, 2017-05-17:
Several prints I have sent to the printer that are 2.5 days freezes on day 2. This happens on both flash drive and USB. Is it because there is not enough memory?

Ghostkeeper, 2017-05-18:
G-code is always streamed in from the USB stick, SD card or USB cable. It's never kept in memory completely.

It's either some bug in the firmware of your printer or a piece of g-code that explicitly pauses it. Do you know where approximately in the g-code it stopped?

partsoven, 2017-05-18:
I'm not sure. It just froze up and wouldn't even let me do anything. I had to restart the UM to lower the bed.

Ghostkeeper, 2017-05-19:
There is no g-code that Cura can output that should cause the printer to freeze entirely (including the screen and such). I'm blaming the firmware! :fire:

Ping to @daid. Also, what printer and firmware version are you using? I assume it's an UM2? Perhaps it warrants a bug report in [Ultimaker2Marlin](https://github.com/Ultimaker/Ultimaker2Marlin/issues).

closed 2018-03-02 by ChrisTerBeke