Ultimaker/Cura #1842: Cura 2.5.0 Non-Beta and Beta crashes randomly on startup on Win10 x64

Fiiti, 2017-05-18:
Hi,
my Cura is randomly crashing during startup. I wonder why. After 2, 3 times laters it starts and I can use them.
I installed the latest nvidia Graphic Driver.
It happens with Windows10 1607 also with 1703. 
This kind of trouble was from the very first day of the installation.

[cura logfiles copy.zip](https://github.com/Ultimaker/Cura/files/1009914/cura.logfiles.copy.zip)

Any suggests, what I can do here?

Regards,
Fiiti

nallath, 2017-05-18:
I don't see anything in the logs unfortunately.

Ghostkeeper, 2017-05-18:
Well there is a 404 error here:

    2017-05-18 08:11:04,775 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Prusa i3, which doesn't exist.
    2017-05-18 08:11:04,776 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine Prusa i3, which doesn't exist.

It shouldn't crash on that though. In fact it seems to continue and suspiciously crashes right after loading the shaders. We're not reproducing this on our test computer with Windows 10, which also has Nvidia.

closed 2018-02-21 by ianpaschal