Ultimaker/Cura #1844: 100% scaling problem

af2334, 2017-05-18:
I'm in Cura 2.5.0 Mac and when I bring a dual color STL model in and merge, the model scale changes even though uniform scaling is on. I get x 100%, y 91.73, z 100%.

It continually happens and if you correct it to all 100%, it actually expands the model in the axes. I have verified dimension and seems to be a glitch.

Any info?

Ghostkeeper, 2017-05-19:
It's not happening when I try it. Does this only happen for certain models? Did you rotate the objects beforehand?

af2334, 2017-05-19:
I double checked it and it seems to happen with merged/dual color prints and indeed shows up when I rotate the model on the bed. The mm dimension stay the same but the percentage shows a smaller number on the Y axis.

AUSTIN FLINK
Creative Design Manager/VizLab
Associate

Building Community for 30 Years

RICE FERGUS MILLER
275 Fifth Street, Suite 100, Bremerton WA 98337
rfmarch.com<http://www.rfmarch.com/>
O  360-377-8773
D  360-362-1420

––––––––––––––––––––––––––––––––––––––––––––––––––

Click Here for Confidentiality Notice & Full Copyright Disclosure<http://www.rfmarch.com/confidentiality.html>

closed 2018-03-02 by ChrisTerBeke