Ultimaker/Cura #1845: Cura 2.5.0 opens multiple instances when importing files from fustion 360 (windows)

lsupl, 2017-05-18:
When trying to import multiple objects from fusion one at a time Cura opens a new instance for each object. This doesn't occur on the previous version that I have used on mac.

Ghostkeeper, 2017-05-19:
Just tried this on Ubuntu and two Windows computers. The --single-instance parameter seems to work fine. But could you try that as well? Just to be sure?

To do that:

* Go into your installation folder `C:\Program Files\` and shift+right-click on the Cura folder there.
* Select "Open in terminal".
* Shift+right-click on the Cura folder again and open a second terminal.
* In the first terminal type `Cura.exe --single-instance`.
* Wait for Cura to start.
* In the second terminal type `Cura.exe --single-instance` as well.

It should not start Cura again the second time.

If it's opening two instances of Cura, we have a bug on our hands in the single-instance code. It should detect the other Cura instance.

fieldOfView, 2017-05-19:
In Fusion 360 you would have to launch Cura with that extra argument. See https://github.com/Ultimaker/Cura/issues/1402#issuecomment-278323500

Drayson80, 2017-05-21:
It does the same (multiple instances) when opening files from e.g. the Explorer (Win10) or download sources like Thingiverse, Youmagine,... Seems there is an instance bug...

Ghostkeeper, 2017-05-21:
It is Cura's default behaviour to create a new instance for every file you open.

...

Perhaps we should change that? You think that would be more useful? Considering the considerable start-up time of the application right now...

Drayson80, 2017-05-22:
I think it should open the files in the same instance - makes it a lot easier to combine files from different sources into one print plate e.g. when opening selected files from a multiple file thingiverse thing.
or make it selectable in configuration...?

godefroi, 2017-06-22:
@fieldOfView Cura 2.5.x supported that extra argument, but Cura 2.6 seems to have lost support for it. Also, it would be super convenient if Cura simply *added* the model to the build plate, instead of *replacing* the build plate with the model (that's how it behaved in 15.x).

iFreilicht, 2017-10-15:
This should be closed as duplicate of #1402, as it is the older ticket.

closed 2017-10-16 by nallath