Ultimaker/Cura #1846: Cura-2.5.0 cannot render beta project

EmilienIT, 2017-05-19:
[image]

Opening all my project saved with Cura-2.5.0-Beta-2 show bad render in Cura-2.5.0 (latest stable).
I don't know if is still a graphics render bug because I don't proceed with print it.
If I open an *.stl file in the latest Cura-2.5.0 there are no render issue.

This is a bug of Cura-2.5.0 Stable or a settings that I can adjust?

Best regards,

EmilienIT

nallath, 2017-05-19:
There is a known issue that the winding order of certain files is not correct. It's not very high on our priority list. If you find it very annoying, you can hide the effect by disabling the display overhangs in your preferences.

EmilienIT, 2017-05-19:
I try with Display overhangs disable but no difference show:
[image]
I take off the option and than restart Cura before take the screenshot.

I hope it will be fixed in the next release ;)

nallath, 2017-05-19:
The red areas are gone, i thought you meant those.

fieldOfView, 2017-05-19:
I think he meant the geometry changes to the body of the butterfly.

EmilienIT, 2017-05-19:
@fieldOfView Yes, I mean the bad resolution.
@nallath Zoom-in the 2nd butterfly and you see the resolution of border is very low.
If I open the same project in Cura-2.5.0-Beta-2 the resolution is high as usual (the same of 1st butterfly from .stl file).
This is a new bug?

nallath, 2017-05-19:
I don't know. I've just tried to save a number of objects to 3mf and load them again, but I don't get the artefacts. 

Could you share the files (stl & 3mf?)

EmilienIT, 2017-05-19:
@nallath : If I save from Cura-2.5.0 (stable version) and load 2mf there are no problems.
The issue comes when I load 3mf that was saved from Cura-2.5.0-Beta-2.
github don't let me attach stl and 3mf files also zip together, I upload here: https://file.io/iSCeaH

Let me know if you download right the zip.

nallath, 2017-05-19:
Ooh! Yes this is a bug that was fixed for the stable release.

closed 2017-05-19 by nallath

EmilienIT, 2017-05-19:
Oh ok, I can open the file saved with Cura-2.5.0-Beta-2 only in Beta-2 version?
Can you provide me the link of Cura-2.5.0-Beta-2 Windows installer?
Because I have a lot of project saved in Beta-2 that I want to print in high resolutions.

Ghostkeeper, 2017-05-21:
I think the file is irreparably damaged. But you can try opening the file in the beta version.

You can download a legacy version of Cura here: https://ultimaker.com/en/products/cura-software/list
Scroll down a bit to find the beta versions. You want the beta for 2.5.

EmilienIT, 2017-05-22:
@Ghostkeeper No, the files 3mf are not damaged because yesterday I open with old Cura-2.5.0-Beta-2.AppImage and I see AND print perfectly.
The same file load on Cura-2.5.0 stable show the same issue, like the screenshots example, the border are all weak.
Seems like Cura-2.5.0 stable is not retrocompatible when loading 3mf from previous release. (I tested only from Cura-2.5.0-Beta-2)

Thank you for your tip on scroll down the page ;)

nallath, 2017-05-23:
They are "damaged" as they do not adhere to the specification of 3mf. I made a mistake as to what locale needs to be used for the way numbers are saved. It should always use the US locale, but instead I specified nothing, which defaults to the user set locale. This means that in some cases decimal numbers are saved like 15,5 instead of 15.5