Ultimaker/Cura #1849: Order sequence of parts in Cura & UM3

engrSM, 2017-05-19:
**Feature Request: the order sequence of parts would be selectable**

I want to print a part using custom-made prime towers. Therefore, I want to print this sequence for each layer:

1. Prime Tower 1 (with Material 1)
2. Part (with Material 1)
3. Prime Tower 2 (with Material 2)
4. Part (with Material 2)

Using per-model-settings, I can easily set the material. However, Cura insists on printing Prime Tower 2 last.

I can shuffle around the plate while watching the first layer travels to make sure the correct sequence occurs, but ideally **the order sequence of parts would be selectable**. If more than a couple parts are on the build plate, it is difficult to force a particular order  This feature would be especially useful when printing PC with PVA supports because PVA cannot easily be deposited on PC, although PC can be deposited on PVA--therefore, PVA supports can be built from the buildplate, but the PC and PVA nozzles must be primed separately.

[image]

[image]

Ghostkeeper, 2017-05-21:
If I recall correctly, Cura has these two requirements for the ordering of objects:
* Minimise the number of extruder switches.
* If given the choice with minimised extruder switches, print the objects in the order in which they are loaded into Cura.

I might be wrong though. A lot of this stuff was moved around when we did the multi-threading of CuraEngine because the layer starting position was no longer known.

engrSM, 2017-05-22:
@Ghostkeeper Thanks for reply. In my experience, I saw no difference in the order of printing when I varied the order in which the models were loaded in Cura (2.5.0).

BagelOrb, 2017-07-21:
implemented this