Ultimaker/Cura #1850: Gcode Flavor UM2+ issue

ChrisRiddell, 2017-05-20:
Hello,

Cura 2.5.0
Ultimaker 2+
Windows 10
Ultimaker firmware was updated when Cura 2.4 came out

If I set the Gcode Flavor to RepRap so I can make use of the temp settings when the printer read the Gcode it has the following issues,

1) When the nozzle moves to the front it over shoots the build plate (goes just passed it)

2) This is a wired issue but when it primes and doing the skirt/brim pretty much no plastic comes out the nozzle, was set to 210c for the first later but no plastic came out, I changed back to the Ultimaker Gcode flavor same file same settings and it primed and extruded perfect. (not sure if this is a firmware issue or a Cura one)

fieldOfView, 2017-05-20:
It sounds like the printer is not homing or heating up.

Have you changed the start gcode at all? How do you send the gcode to the printer? Via USB cable or via SD-card? Note that with Reprap (Sprinter/Marlin) gcode it should not matter, but I would like to know anyway.

Can you post a gcode file somewhere?

ChrisRiddell, 2017-05-20:
Hi @fieldOfView 

I did not pay much attention if it done a home before it moved but it came so close to bashing into the front, it shows on screen that it heated up. (everything was fine under the Ultimaker flavor)

1) Nope the start Gcode is what Cura has set
[image]

2) It was printed using SD-card.

Here is the .gcode file
[UM2_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/1016034/UM2_3DBenchy.zip)

fieldOfView, 2017-05-20:
Looking at the gcode, I don't see any strange coordinates that would cause the printer to try to move to unprintable areas.

>  it came so close to bashing into the front

But it did not actually bash into the front? The default start gcode if you use Reprap flavor is to prime the nozzle in the front of the buildplate. If you don't want it to do that, remove the line that says ```G1 X15 Y0 F4000 ;move X/Y to front of printer```

ChrisRiddell, 2017-05-20:
It did not bash the front just came close, but that makes sense what it done if it primes infront of the build plate.

So number 1 is ruled out as a bug just the heating/extruding does the Reprap flavor have the same prime procedure? 

Could it be possible that the Reprap priming is not enough to get things going as I stopped the print when pretty much nothing came out from the prime and nothing for the skirt (only a small model so the skirt is not big)

fieldOfView, 2017-05-20:
> when it primes and doing the skirt/brim pretty much no plastic comes out the nozzle

You may want to prime a bit more then. Change the ```G1 F200 E10 ; extrude 10 mm of feed stock``` line to say E16 instead of E10 to prime 6 mm more.

fieldOfView, 2017-05-25:
This issue can be closed, right?

ChrisRiddell, 2017-05-26:
Have not had time to test this I'll do it in the morning as it's a bit late here now.

ChrisRiddell, 2017-05-31:
Changing to E16 fixed it, tested using Ultimaker filament and Polymaker filament.

ChrisRiddell, 2017-06-02:
@fieldOfView to get a useful prime (still way less then ulticode) but any less even with a skit on a small model is not enough :/ 

`G1 F200 E50 ; extrude 10 mm of feed stock`

closed 2017-07-06 by ChrisRiddell