Ultimaker/Cura #1852: No support when surface mode = surface

titatovenaar1, 2017-05-20:
When setting the surface mode to surface (within special modes) no support is generated. My printer is an UM3.

greets,

BagelOrb, 2017-05-22:
Why would you want support and surface mode? The surface will then probably stick more to sort than to other surface lines.

titatovenaar1, 2017-05-22:
I want to print an automotive bumper part in scale 1/10. These parts are real CAD data for new projects The walls are very thin and only show in surface mode

BagelOrb, 2017-05-22:
I suggest you set the Horizontal Expansion to something like 0.2mm instead.

<https://github.com/Ultimaker/Cura/issues/1852#issuecomment-303131178>, or mute
the thread

titatovenaar1, 2017-05-22:
I tried the site MakePrintable. This seems to work pretty good.

Thanks, I will try the horizontal expansion

titatovenaar1, 2017-05-22:
Hi I tried the horizontal expansion, but the support disappeared same as with the surface mode.

BagelOrb, 2017-05-22:
That sounds odd. Are you sure you hadn't left the surface mode on by
accident? Are you sure the model even needs support?

titatovenaar1, 2017-05-23:
Ok, problem is solved with the horizontal expesion set to .2

titatovenaar1, 2017-05-23:
Expansion

closed 2017-05-23 by BagelOrb