Ultimaker/Cura #1854: Cura 2.5, TweakAtZ 5.1: Tweak Print Speed thows an exception

anton-veretenenko, 2017-05-21:
Log:
```
2017-05-22 00:37:44,227 - ERROR - UM.Logger.logException [74]: Exception: Exception in post-processing script.
2017-05-22 00:37:44,230 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-05-22 00:37:44,233 - ERROR - UM.Logger.logException [78]:   File "/tmp/.mount_qlT5sR/usr/bin/plugins/plugins/PostProcessingPlugin/PostProcessingPlugin.py", line 62, in execute
2017-05-22 00:37:44,235 - ERROR - UM.Logger.logException [78]:     gcode_list = script.execute(gcode_list)
2017-05-22 00:37:44,237 - ERROR - UM.Logger.logException [78]:   File "/tmp/.mount_qlT5sR/usr/bin/plugins/plugins/PostProcessingPlugin/scripts/TweakAtZ.py", line 429, in execute
2017-05-22 00:37:44,240 - ERROR - UM.Logger.logException [78]:     modified_gcode += "G1 F%d X%1.3f Y%1.3f E%1.5f\n" % (int(f/100.0*float(printspeed)),self.getValue(line,"X"),
2017-05-22 00:37:44,242 - ERROR - UM.Logger.logException [78]: NameError: name 'printspeed' is not defined
```

But Tweak Speed value works though.

Ghostkeeper, 2017-05-22:
Looks like that `printspeed` should be replaced by the value of the `f2_printspeed` setting. I don't have the tools to test it here, but I'm making a note to check tomorrow.

Ghostkeeper, 2017-05-22:
Made a small fix for you: https://github.com/nallath/PostProcessingPlugin/commit/a4f16872d18ebc6214be696113fa35636ab1860e

It seems to work again now. Thanks for the report!

closed 2017-05-22 by Ghostkeeper