Ultimaker/Cura #1855: Cura 2.5, TweakAtZ 5.1: layers count does not match Cura layers view mode

anton-veretenenko, 2017-05-21:
In "View Mode: Layers" Cura counting layers starting from 1. Gcode files are generated with layers counting from 0. And TweakAtZ also count layers starting from 0. So it is always a miss if you enter a layer number from Cura layers view mode into TweakAtZ instance.

Ghostkeeper, 2017-05-22:
Perhaps subtracting 1 from the layer indicated in TweakAtZ would suffice.

anton-veretenenko, 2017-05-22:
yes, but TweakAtZ accepting layer 0, at least for now.

Ghostkeeper, 2017-05-22:
I see. It actually looks for the `;LAYER:` comments in the code and bases it on that. In fact it accepts layer numbers down to -100. These are the raft layers, which have negative layer numbers.

Ghostkeeper, 2017-09-04:
The pull request over at https://github.com/Ultimaker/Uranium/pull/258 mentions that it would fix this bug, but it doesn't. It has nothing to do with this issue.

Ghostkeeper, 2017-09-04:
The ticket for this bug was just removed from our planning by our project manager for being more than 3 months old.

closed 2023-06-24 by github-actions[bot]