Ultimaker/Cura #1856: Cura 2.5, Post Processing Plugin: gcode does not re-postprocess after changing scripts values

anton-veretenenko, 2017-05-21:
Slice automatically disabled.

I've sliced model with "Prepare". Then I've added some TweakAtZ instances or other Post Processing scripts. Then used "Save to File" button to save gcode. Gcode appeared post-processed.
Then I go and change some values in tweaks/scripts and use "Save to File" again (no "Prepare" button appear). Gcode file appeared as previous one with previous tweaks.
This is from log:
`
2017-05-22 01:01:08,777 - ERROR - PostProcessingPlugin.PostProcessingPlugin.execute [69]: Already post processed
`

If I force some changes in slicing options to make "Prepare" button appear then it re-postprocess gcode and new tweak values appear in gcode.

Ghostkeeper, 2017-08-20:
It reproduces for me, but our product manager just closed this issue off for being more than 3 months old.

d235j, 2017-10-28:
Filed upstream — this looks like a bug in the PostProcessing plugin.

d235j, 2017-10-30:
Hmm, doesn't seem to reproduce anymore for me.

closed 2017-10-30 by nallath

Ghostkeeper, 2017-10-30:
For me neither. It seems to have been fixed in the meanwhile.