Ultimaker/Cura #1857: Cura 2.5 No manual control means I cannot change my filament

bkofford, 2017-05-22:
I have filament I cannot use in my hotend. I need to heat up the hotend in order to remove it and insert a different filament. In older versions of Cura, this was trivial, but now I find it impossible, because there are no manual controls that allow me to instruct my Printrbot Play to heat up the hotend to allow me to change the filament.
WHY?

nallath, 2017-05-22:
Because continuing to support USB printing took us quite a bit of time, so at some point tough choices had to be made. As hardly any ultimaker printer uses USB printing, we decided to no longer put any effort into it.
Others are working on it. But if you want your PrintrBot to work well with Cura, you might want to ask them to also put some effort into supporting Cura.

closed 2017-05-22 by nallath

Ghostkeeper, 2017-05-22:
This was left out when the front-end was rewritten from scratch at version 15.06.