Ultimaker/Cura #1858: Fatal Execption occurred on startup which reoccurs when after reboot and try again

ConradWarwick, 2017-05-22:
Version: 2.3.1
Platform: Windows-10-10.0.15063
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

ConradWarwick, 2017-05-22:
Reinstall of Robo 2.3.1 resolved issue

closed 2017-05-22 by ConradWarwick

Ghostkeeper, 2017-05-24:
It's trying to load your log file, which has grown too big to contain in Python's memory. This was solved in 2.4.