Ultimaker/Cura #1862: No way to put Relative extrusion

Benraay, 2017-05-22:
Hi everybody,

Thanks for the work you all do on this Slicer.
I have a printer that does not support Absolute extrusion, I need absolutely to output Gcode with Relative E values, I searched for a long time and Googled but nothing on how to change that setting in the Machine settings can someone help ?

nallath, 2017-05-23:
Not as far as I know. @BagelOrb

BagelOrb, 2017-05-23:
It's not possible and we don't need it for our machines so we're not going to implement it.

closed 2017-05-23 by BagelOrb

Ghostkeeper, 2017-05-24:
This could be solved with a post-processing script, perhaps. Basically you need to go over every line and see how much extrusion is used and subtract the previous E-value from the current extrusion.

Benraay, 2017-05-24:
yes it's what I planned to do, looks not so difficult.

Benjamin Raaymakers

CONFIDENTIALITY NOTICE:

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message has
been addressed to you in error, please immediately alert the sender by
reply email and thendelete this message and any attachments. If you are not
the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments
is strictly
prohibited.