Ultimaker/Cura #1867: Question: printing direction of walls

maukcc, 2017-05-23:
Is there a way to print the walls in alternate directions.
So if you are printing a circle, one layer goes CW and the next CCW.

nallath, 2017-05-24:
I don't think so. @BagelOrb?

BagelOrb, 2017-05-24:
Nope, why would you want that?

maukcc, 2017-05-24:
We are printing with viscous materials, and the layers do not get "hard" like thermoplastics.
The previous layers are getting dragged (very little) in the direction that the nozzle is going.
By alternating the direction on walls, we wanted to see if we can compensate for this dragging.

BagelOrb, 2017-05-24:
In that case I think you need to do some tests first.
Maybe you can test by manually changing a gcode file and see what the
results are like.

maukcc, 2017-05-29:
This can be closed.

BagelOrb, 2019-01-26:
All layers are already going in the same direction. There's no reason to have a user setting which let's you do otherwise.

MB3Y, 2019-01-26:
Sorry if I didn't make myself clear: I meant: the directions of inner and outer walls in a certain layer. Some go CW, others CCW. In between layers they are aligned indeed

MB3Y, 2019-01-26:
In fact it's not even always about bleeding on my diamond hotend printer. Also Prusa i3 MK3 with translucent PETG has some inconsistencies I can't get rid of and which I attribute to the fact that the material has less time to cool at the point where the change of direction happens...

Ghostkeeper, 2019-02-14:
Outside walls should always go CCW (also inner walls for outside walls). Walls for holes in your layer should always go CW. If the behaviour is different, could you share a project file that reproduces the problem?

MB3Y, 2019-02-17:
Thanks @Ghostkeeper to reach out. 
1. Cooling time:
The problem I've experienced the other day is when using translucent PETG and not using infill (which was pretty logical here, as objectwalls are thin enough. The material has plenty of time to cool down for CW or CCW outer and innerwalls, but where the inner walls from the inside and the outside object walls meet you have the  directional change. CW innerwall completes, changes direction for CCW inner wall and there you have one walls passing along a wall that hasn't got much possibility to cool as much as the rest of the wall...
2. Question: what is the reason for CCW and CW visible walls in one object? This direction change makes another bleeding effect, since one filament color runs over the other (e.g. white over black makes more white on outerwall visible), but changing direction makes black run over white, making black more visible on the edges... So for me CW and CCW give different mixing/bleeding effects.
For objects like the one in attachment there are only walls and my poor minds basic reflex is to print all walls in the same direction like a 4walled spiralized contour (without smoothing of course)?

[image]
[Box.zip](https://github.com/Ultimaker/Cura/files/2873259/Box.zip)

Ghostkeeper, 2019-02-18:
>     2\. Question: what is the reason for CCW and CW visible walls in one object?

This is an implementation artefact. The library that we're using for Polygon operations (Clipper) stores vertices of the polygon in this order, and we just iterate over it. We can consider iterating over it in reverse direction for walls of holes, but that may not be simple because there is a bunch of stuff going on with the Z-seam and travel move optimisation there, and of course you have to detect that the current wall is a wall for a hole.

Ghostkeeper, 2019-02-18:
We want to make sure that these walls are printed in the same direction, we've agreed.

jsourice, 2020-06-05:
Julien Sourice, from Nanoe french company. We 3D print ceramic and metal part on any FDM.
[https://nanoe.com/en/](url)

I would be very interested in such feature as alternating printing direction of walls CW/CCW every layer. I asked if this option exists or could be implemented in PrusaSlicer here : 

[https://github.com/prusa3d/PrusaSlicer/issues/4263](url)

You'll find also in this issue more information about Nanoe and why this feature can be pretty critical for our ceramic application. Direct modification of the gcode does work for small Z axis part, but it is realy not user-friendly.

BagelOrb, 2020-06-05:
The reason we asked to do it by hand is because we would like to see pictures first before we are convinced that this should be a feature in Cura. Without any proof that this setting is required in any context (eg ceramic printing) we would not want to have an extra setting for this.

I already have a feature implemented which alternates the direction of walls accross layers, but also between the walls within one layer. However, I am not yet convinced that it helps for the use-case I am currently working on. Therefore it is not yet merged into Cura.

We first need proof that this feature actually improves a certain printing setup (printer+material+nozzle).

BagelOrb, 2020-06-05:
Never mind. It's already under consideration for including in Cura: https://github.com/Ultimaker/CuraEngine/pull/1250

Hopefully the functionality will soon be merged and available to all users.

jsourice, 2020-06-05:
Ok ! I was about to repply that we can do a use case to prove that this would enhance drasticaly our parts quality. I guess now we just have to be patient ! Thank you.

BagelOrb, 2020-06-05:
If you include a use-case the team might be inclined to deal with the PR faster. Sometimes it takes up to 6 months to process a PR, so you might want to help to get it done faster.

closed 2020-11-26 by nallath