Ultimaker/Cura #1868: Cura 2.6.0 beta - crash when selecting customized printer profile

Drayson80, 2017-05-23:
the printer profiles as well as extruder profiles work on 2.5 without issue. 
The printer has been added also with crash too, but after a new start it was in the list. As soon as I select the printer, Cura crashes with the following report:

Version: 2.6.0-BETA
Platform: Windows-7-6.1.7601-SP1
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 491, in _onStackChanged
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\BuildVolume.py", line 555, in _updateDisallowedAreas
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderManager.py", line 459, in getUsedExtruderStacks
KeyError: '0'

-------------------------
Thread 0x00001748 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 150 in _storeStderrToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001830 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 142 in _storeOutputToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000116c (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000013b0 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001ce4 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001410 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001cbc (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000648 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001b20 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000000d8 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001080 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000011f0 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000015f4 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000122c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001ee8 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000b50 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x00001164 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\ExtruderManager.py", line 514 in __globalContainerStackChanged
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315 in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212 in emit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Application.py", line 152 in setGlobalContainerStack
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 346 in setActiveMachine
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 659 in run
  File "D:/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

nallath, 2017-05-24:
Ugh. This again. I thought we managed to fix it.. :( 

Could you share your entire log?

Drayson80, 2017-05-24:
Maybe it's just me :-)
I could share, but it is app. 12Mb with a lot of stuff since 2016-09-02. How shall I?

nallath, 2017-05-24:
Pastebin?

Drayson80, 2017-05-24:
lets try this way: https://we.tl/39Y2dxvIo1 - should be up for app. 10 days...

Ghostkeeper, 2017-05-30:
Perhaps this is the relevant part of that log:

    2017-05-24 11:56:20,271 - DEBUG - CuraEngineBackend.ProcessSlicedLayersJob.run [228]: Processing layers took 0.320453405380249 seconds
    2017-05-24 11:57:24,470 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine UMO SiRight Extrusion CE, which doesn't exist.
    2017-05-24 11:57:24,470 - WARNING - cura.Settings.ExtruderManager.getMachineExtruders [404]: Tried to get the extruder trains for machine UMO SiRight Extrusion CE, which doesn't exist.
    2017-05-24 11:57:24,758 - WARNING - cura.Settings.SettingInheritanceManager.getOverridesForExtruder [55]: Unable to find extruder for current machine with index 0
    2017-05-24 11:57:24,760 - WARNING - cura.Settings.SettingInheritanceManager.getOverridesForExtruder [55]: Unable to find extruder for current machine with index 0
    2017-05-24 11:57:24,762 - WARNING - cura.Settings.SettingInheritanceManager.getOverridesForExtruder [55]: Unable to find extruder for current machine with index 0
    2017-05-24 11:57:24,764 - WARNING - cura.Settings.SettingInheritanceManager.getOverridesForExtruder [55]: Unable to find extruder for current machine with index 0
    2017-05-24 11:57:24,766 - WARNING - cura.Settings.SettingInheritanceManager.getOverridesForExtruder [55]: Unable to find extruder for current machine with index 0
    ...

Ghostkeeper, 2017-05-30:
I also found this a bit further back:

    2017-05-24 11:42:21,797 - DEBUG - UM.Settings.ContainerStack.deserialize [285]: While deserializing, we got the following container string: ultimaker_original_dual_2nd_CE #2_current_settings,umo dual extrusion ce_normal quality #2,normal,mexhibit _PVA_support_white,empty_variant,ultimaker_original_dual_2nd_CE,
    2017-05-24 11:42:21,812 - ERROR - UM.Logger.logException [74]: Exception: Could not deserialize container ultimaker_original_dual_2nd_CE #2
    2017-05-24 11:42:21,812 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
    2017-05-24 11:42:21,812 - ERROR - UM.Logger.logException [78]:   File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Settings\ContainerRegistry.py", line 194, in load
    2017-05-24 11:42:21,812 - ERROR - UM.Logger.logException [78]:   File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Settings\ContainerStack.py", line 294, in deserialize
    2017-05-24 11:42:21,812 - ERROR - UM.Logger.logException [78]: Exception: When trying to deserialize ultimaker_original_dual_2nd_CE #2, we received an unknown ID (ultimaker_original_dual_2nd_CE #2_current_settings) for container
    2017-05-24 11:42:21,812 - DEBUG - UM.Settings.ContainerStack.deserialize [285]: While deserializing, we got the following container string: ultimaker_original_dual_1st_CE #2_current_settings,umo dual extrusion ce_normal quality #2,normal,Colorfabb_PLA_generic,empty_variant,ultimaker_original_dual_1st_CE,
    2017-05-24 11:42:21,828 - ERROR - UM.Logger.logException [74]: Exception: Could not deserialize container ultimaker_original_dual_1st_CE #2
    2017-05-24 11:42:21,828 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
    2017-05-24 11:42:21,828 - ERROR - UM.Logger.logException [78]:   File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Settings\ContainerRegistry.py", line 194, in load
    2017-05-24 11:42:21,828 - ERROR - UM.Logger.logException [78]:   File "D:/2.5/build/inst/lib/python3.5/site-packages/UM\Settings\ContainerStack.py", line 294, in deserialize
    2017-05-24 11:42:21,828 - ERROR - UM.Logger.logException [78]: Exception: When trying to deserialize ultimaker_original_dual_1st_CE #2, we received an unknown ID (ultimaker_original_dual_1st_CE #2_current_settings) for container
    2017-05-24 11:42:21,843 - DEBUG - UM.Settings.ContainerRegistry.load [204]: Loading data into container registry took 9.253697872161865 seconds

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
We've had a lot of these for the 3.0 release. They seem to have been fixed for the 3.1 release of Cura.