Ultimaker/Cura #1869: Cura 2.5 & 2.6 - Processing Layers 99% (Never finishes)

jatmn, 2017-05-23:
I have not been able to do layer preview with Cura 2.5 or 2.6
Selecting the option causes a "Processing Layers 99%" dialog to popup but then just stays there and it never (even after 5hrs) shows the layers preview. Dragging the slider, changing checkboxes and view mode in the popup window results in no change of state. 

I am honestly not sure where to go with troubleshooting this. 

Windows 8.1

fieldOfView, 2017-05-23:
> (even after 5hrs)
Now that's dedication!

What OS, what GPU, do the logs say anything?

jatmn, 2017-05-23:
Here is some log data. Noticed it calling my D:/ which is odd.. that's my SD card reader. 
Wonder if that's related to a loading issue I have as well? I can't launch Cura without a SD card loaded because it hangings trying to access the SD card and wont finish loading until I put one in.
OS: Windows 8.1 
GPU: AMD R9 390

2017-05-23 14:01:43,706 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [474]: Slicing took 1.6243312358856201 seconds
2017-05-23 14:01:43,709 - DEBUG - UM.Backend.Backend._logSocketState [176]: Socket state changed to Closing
2017-05-23 14:01:43,713 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-05-23 14:01:43,715 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-05-23 14:01:43,732 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [585]: Backend quit with return code 0. Resetting process and socket.
2017-05-23 14:01:44,322 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to LayerView
2017-05-23 14:01:44,325 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 2.6\plugins\LayerView\layerview_composite.shader]...
2017-05-23 14:01:44,577 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 2.6\resources\shaders\color.shader]...
2017-05-23 14:01:44,584 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 2.6\plugins\LayerView\layers3d.shader]...
2017-05-23 14:01:44,590 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 2.6\resources\shaders\toolhandle.shader]...
2017-05-23 14:01:44,608 - ERROR - UM.Logger.logException [76]: Exception: Job <CuraEngineBackend.ProcessSlicedLayersJob.ProcessSlicedLayersJob object at 0x0000002D525566D8> caused an exception
2017-05-23 14:01:44,613 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-05-23 14:01:44,616 - ERROR - UM.Logger.logException [80]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 125, in run
2017-05-23 14:01:44,618 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.6\plugins\CuraEngineBackend\ProcessSlicedLayersJob.py", line 176, in run
2017-05-23 14:01:44,618 - ERROR - UM.Logger.logException [80]:     color = colorCodeToRGBA(color_code)
2017-05-23 14:01:44,621 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.6\plugins\CuraEngineBackend\ProcessSlicedLayersJob.py", line 35, in colorCodeToRGBA
2017-05-23 14:01:44,623 - ERROR - UM.Logger.logException [80]:     int(color_code[1:3], 16) / 255,
2017-05-23 14:01:44,625 - ERROR - UM.Logger.logException [80]: TypeError: 'NoneType' object is not subscriptable
2017-05-23 14:01:51,577 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

Ghostkeeper, 2017-05-24:
This occurs when a material has no metadata entry "color".

I fixed this here: 3ff9ad5d4e917af9ef1c0ff2c0bcaaea46c0d8f5.

closed 2017-05-24 by Ghostkeeper

doc1623, 2017-10-15:
It appears this issue or a closely related one still exist. I get this exact error but from another missing attribute. 

2017-10-15 10:33:58,891 - ERROR - UM.Logger.logException [76]: Exception: Job <CuraEngineBackend.ProcessSlicedLayersJob.ProcessSlicedLayersJob object at 0x7f0019c56ac8> caused an exception
2017-10-15 10:33:58,891 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-10-15 10:33:58,892 - ERROR - UM.Logger.logException [80]:   File "/usr/lib64/python3.5/site-packages/UM/JobQueue.py", line 125, in run
2017-10-15 10:33:58,892 - ERROR - UM.Logger.logException [80]:     job.run()
2017-10-15 10:33:58,893 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/CuraEngineBackend/ProcessSlicedLayersJob.py", line 178, in run
2017-10-15 10:33:58,893 - ERROR - UM.Logger.logException [80]:     color_code = material.getMetaDataEntry("color_code", default="#e0e000")
2017-10-15 10:33:58,893 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'
2017-10-15 10:34:04,280 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

My real problem is another issue entirely. I can't create a custom material. It can't find a needed module

2017-10-15 10:55:53,498 - ERROR - UM.Logger.logException [80]: ImportError: No module named 'zeroconf'
2017-10-15 10:55:53,499 - ERROR - UM.PluginRegistry._populateMetaData [323]: Could not find plugin UM3NetworkPrinting

So without a material, I get the exact same issue jatmn had.

doc1623, 2017-10-15:
I imported a material profile from 

`https://github.com/Robo3D/Slicing-Profiles/tree/master/Cura%202/Material%20Profiles` 

and I still get this issue

2017-10-15 11:45:44,840 - ERROR - UM.Logger.logException [76]: Exception: Job <CuraEngineBackend.ProcessSlicedLayersJob.ProcessSlicedLayersJob object at 0x7f6a95ea1668> caused an exception
2017-10-15 11:45:44,840 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-10-15 11:45:44,841 - ERROR - UM.Logger.logException [80]:   File "/usr/lib64/python3.5/site-packages/UM/JobQueue.py", line 125, in run
2017-10-15 11:45:44,841 - ERROR - UM.Logger.logException [80]:     job.run()
2017-10-15 11:45:44,841 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/CuraEngineBackend/ProcessSlicedLayersJob.py", line 178, in run
2017-10-15 11:45:44,842 - ERROR - UM.Logger.logException [80]:     color_code = material.getMetaDataEntry("color_code", default="#e0e000")
2017-10-15 11:45:44,842 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

doc1623, 2017-10-15:
ok weird. I changed to another profile and changed the color. It says it's successfully imported but I still get the same message. The color should be red...  #ff0000

2017-10-15 11:53:39,663 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [474]: Slicing took 3.38330340385437 seconds
2017-10-15 11:53:39,664 - DEBUG - UM.Backend.Backend._logSocketState [176]: Socket state changed to Closing
2017-10-15 11:53:39,664 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-10-15 11:53:39,664 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-10-15 11:53:39,665 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [585]: Backend quit with return code 0. Resetting process and socket.
2017-10-15 11:53:43,361 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to LayerView
2017-10-15 11:53:43,362 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/lib/cura/plugins/LayerView/layerview_composite.shader]...
2017-10-15 11:53:43,775 - ERROR - UM.Logger.logException [76]: Exception: Job <CuraEngineBackend.ProcessSlicedLayersJob.ProcessSlicedLayersJob object at 0x7f2b0e1e4160> caused an exception
2017-10-15 11:53:43,776 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-10-15 11:53:43,776 - ERROR - UM.Logger.logException [80]:   File "/usr/lib64/python3.5/site-packages/UM/JobQueue.py", line 125, in run
2017-10-15 11:53:43,776 - ERROR - UM.Logger.logException [80]:     job.run()
2017-10-15 11:53:43,777 - ERROR - UM.Logger.logException [80]:   File "/usr/lib/cura/plugins/CuraEngineBackend/ProcessSlicedLayersJob.py", line 178, in run
2017-10-15 11:53:43,777 - ERROR - UM.Logger.logException [80]:     color_code = material.getMetaDataEntry("color_code", default="#e0e000")
2017-10-15 11:53:43,777 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'
2017-10-15 11:53:43,779 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/color.shader]...
2017-10-15 11:53:43,783 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/lib/cura/plugins/LayerView/layers3d.shader]...
2017-10-15 11:53:43,819 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/share/uranium/resources/shaders/toolhandle.shader]...
2017-10-15 11:53:46,289 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

Ghostkeeper, 2017-10-22:
The error you're getting indicates that the material currently being active is `None`. It's quite different from the original issue.

Could you try this out with the 3.0 release? We made a few things safer in the way that it finds these materials. The line in question now says this:

    color_code = extruder.material.getMetaDataEntry("color_code", default=default_color)

So it obtains the material directly from the extruder.

commit 3ff9ad5d (2017-05-24): Add default for colour code of material

Otherwise you get an error that the colour code could not be found.

Fixes #1869.
commit ce6814d4 (2017-05-24): Add default for colour code of material

Otherwise you get an error that the colour code could not be found.

Fixes #1869.