Ultimaker/Cura #1870: can't focus submenu with Cura 2.3.1

mokusolo, 2017-05-24:
I searched issues with key words such as "can't focus", "menu" etc. But seems it's not this one.
I have met the problem before when using 2.1 with two different computers, so maybe it's irrelative with whether it's win8 or win10.
Most time when I want to click on submenu such as "Show Changelog"/"Check for updates", it didn't get focused. 
I wonder if anyone else has this problem? Does it have something to do with how plugins worked?
As following screenshot shows, my cursor is actually in the red circle, and it just can't be focused.
[image]
Any reply will be appreciated. Thank you.

fieldOfView, 2017-05-24:
Can you try a newer version, eg the freshly released 2.6 beta? 

We switched to a newer version of Qt a while back. This sounds like an issue in Qt, and it may have been fixed since the version that shipped with 2.3.1

Ghostkeeper, 2017-05-24:
Perhaps try the arrow keys as a workaround.

mokusolo, 2017-05-24:
I run 32bit OS in my virtual machine. And I know it's not supported since Cura 2.4. When I fetch a 64bit OS tomorrow, I will try 2.6.
My friend told me to use arrow keys too. Still not used to that but it will be.
Thank you.

closed 2017-05-25 by mokusolo