Ultimaker/Cura #1871: Default material_print_temperature used in Start Gcode when material selected

tcm0116, 2017-05-24:
I'm using a Start Gcode routine that contains the line:

`M109 S{material_print_temperature}`

If a quality is selected that does not change the `material_print_temperature` and then a material is selected that does change the `material_print_temperature`, the resulting gcode generated by Cura 2.5 will use use the default value of 210 from fdmprinter.def.json in the line above and not the temperature defined by the selected material. This can be problematic for high-temp materials like ABS, since 210 may not be high enough to extrude.

tcm0116, 2017-05-24:
Just checked the beta of 2.6, and the issue still exists. In both versions, if I right click in the Printing Temperature box and select "Copy value to all extruders", then the gcode is generated correctly with the only change being the display of the _i_ icon next to Printing Temperature.

Ghostkeeper, 2017-05-24:
Materials can't change `material_print_temperature`. They can only change `default_material_print_temperature`. `material_print_temperature` should take the value of `default_material_print_temperature` unless overridden by a quality profile.

tcm0116, 2017-05-24:
Looking into this some more, it may be an issue with multi-extruder Custom FDM Printers. The steps below will replicate the issue I'm seeing.

1. Install Cura 2.6 on virgin Windows 10 system
2. Create Custom FDM Printer with 2 extruders and a heated bed
3. Add `M109 S{material_print_temperature}` after G28 line in Start GCode
4. Load a STL file
5. Change material to ABS for both extruders
6. Verify Printing Temperature is 230 for both extruders
7. Save to File
8. Open generated gcode file and see that `M109 S210` has been inserted instead of the expected `M109 S230`

fieldOfView, 2017-05-25:
This does not seem to be limited to Custom FDM Printers, but any multiextruder printer that has replacement patterns that should be retreived from an extruder instead of the global stack.

Token replacement is done in StartSliceJob._buildGlobalSettingsMessage which only gets setting values from the global stack, so it simply ignores values from extruder stacks. This is going to need some rethinking.

tcm0116, 2017-05-25:
@fieldOfView If I right-click in the Printing Temperature window and select copy to all extruders, then the generated GCode has the correct temperature in the M109 line. I assume this action moves the setting to the global stack, which would then correspond to your deduction.

fieldOfView, 2017-05-25:
Yes, sounds logical

Ghostkeeper, 2017-05-28:
Sounds like a duplicate of #1296 then.

closed 2017-05-28 by Ghostkeeper