Ultimaker/Cura #1873: Successful imported profile not showing up

ChrisRiddell, 2017-05-24:
Ultimaker 2+
Windows 10
Cura 2.6 Beta

When I try and import a profile it says "Successfully imported profile profile_name" but it does not show in the list.

Spent a little while stuffing around and found changing 
`[metadata]
quality_type = draft`
to
`[metadata]
quality_type = fast`

Allowed it to be imported so I'm guess something there is some issue importing draft quality types or under a UM2+ machine you can't import a draft type :/ if you can't import a draft type for a UM2+ maybe it should not say it was Successfully imported but give a warning saying its not supported.

fieldOfView, 2017-05-24:
Please post the profile you are trying to import somewhere, so we can have a look.

ChrisRiddell, 2017-05-24:
[Draft.zip](https://github.com/Ultimaker/Cura/files/1024853/Draft.zip)

nallath, 2017-05-24:
Hmm. I wonder how you managed to export a "quality" profile. It should only allow for "quality_changes" to be exported. Importing is designed to work for quality_changes, but it seems to not fail properly if you use it on something else.

ChrisRiddell, 2017-05-24:
@nallath This profile was sent to me by a friend so not sure if he exported it or made it by hand in an editor but changing the type to `type: quality_changes` made no difference still does not show if imported only works if i change the quality_type to fast.

I had a look over some of the profile's in the github and there is ones with a draft quality_type.

awhiemstra, 2017-05-24:
So I just had a look at the profiles for the UM2+. It turns out that we have some profiles that specify their type as "draft" and some that specify it as "fast". Since the exported material came from a machine that does not specify machine-specific profiles, it was based off the global "Draft" profile.

On import, we do set the right machine and material as far as I know, but we assume the quality type to be correct. Since we now have non-matching quality types, the profile would never show up since there is no "draft" type for PLA 0.4 for example.

Ghostkeeper, 2017-05-24:
We just added a warning message to be shown when this happens.

noderat, 2017-05-29:
Looks like I'm running into a similar issue, but I don't seem to have the same setting in my profile. The profile was imported from GCode using Cura 15.04.6, then exported and does not appear in Cura 2.6 beta. I've included the gcode, exported profile and cura.log in the below zip.

[TempTowerCuraProfile.zip](https://github.com/Ultimaker/Cura/files/1036588/TempTowerCuraProfile.zip)

Ghostkeeper, 2017-05-30:
The warning message wasn't added on time for the 2.6 beta.

closed 2017-07-06 by ChrisRiddell